NordicPlayground/nRF51-ble-bcast-mesh #2: Writing to flash

mrquincle, 2015-01-23:
Writing to flash at some time (not that you do in this example!) will lead to the code ending up in the `ts_sd_event_handler` with the `NRF_EVT_FLASH_OPERATION_SUCCESS` event type.

This subsequently leads to the `default` case and `APP_ERROR_CHECK(NRF_ERROR_INVALID_STATE);` will stop execution.

I would recommend to add the clause:

    case NRF_EVT_FLASH_OPERATION_SUCCESS:                                                                       
                break; 

to the code. Also the default one you might want to return actually:

`APP_ERROR_CHECK(evt);`

If you want me to submit a code request, feel free to ask.

trond-snekvik, 2015-01-23:
That's a good point. We've discussed this a bit, and believe that the entire event handling system needs a redesign, and that this is a sign of a bigger issue. The event handling is sort of backwards the way it is today, as the user application calls the event handler, and all sd-events are consumed in the framework. We'll probably turn this around, by having the framework take direct control of the SD ISR, and create some event distribution system where we filter out the events that are relevant to the mesh framework, and send the rest to the application. 

For a more immediate solution your suggestion is a good fit. I believe the default case should still be `APP_ERROR_CHECK(NRF_ERROR_INVALID_STATE);`, as the `APP_ERROR_CHECK()` is to be treated as an assert for nRF51 SDK status codes, rather than checking the validity of SD events.

mrquincle, 2015-01-23:
I agree with the check. You're right. Although for now the user might find out other corner cases faster if the actual error code isn't shielded. 

Looking forwards to the new implementation! I think this would be indeed nice from the viewpoint of the application.

trond-snekvik, 2015-03-03:
The new release includes [a document](../blob/master/docs/integrating_w_SD_apps.adoc) that steps through the necessary steps required for integrating the mesh with existing Softdevice-based applications. Unfortunately, the event handler restructure didn't make it for this release, but we hope that the solution for future problems with this is properly explained in the guide. I'll leave the issue as pending.

trond-snekvik, 2015-08-06:
Hi again,
With the SD8-support, the project now utilizes the SDK-module softdevice_handler, which addresses this issue. The softdevice_handler provides an interface to set different interrupt handlers for the different event classes, which should give the application the needed freedom to propagate events the right way. There are no additional changes to this under the hood in the rbc-mesh, but I think the handler should provide a sufficient interface to handle this case :)

mrquincle, 2015-08-06:
Nice! I will check it out the coming weeks!

closed 2016-02-05 by daviddedwin