NordicPlayground/nRF51-ble-bcast-mesh #3: Errors generated by `sd_ble_gatts_hvx`

mrquincle, 2015-01-23:
Currently, I've two types of errors that are generated by `sd_ble_gatts_hvx` in the function `mesh_srv_char_val_set` when I run your code along side code that publishes services as in your `BLE_Gateway` example.

Those are:

    BLE_ERROR_GATTS_SYS_ATTR_MISSING
    NRF_ERROR_SVC_HANDLER_MISSING

If I just return `NRF_SUCCESS` for these two cases, the code runs (with some hiccups) in parallel with those aforementioned services.

Does this have to do with interplay between these two ways of using the radio? Sorry, that this is such a vague report.

trond-snekvik, 2015-01-26:
For the first error code, try doing a call to `sd_ble_gatts_sys_attr_set(conn_handle, NULL, 0)` in the event handler, like in the nRF51 SDK example "ble_app_pwr_profiling" (main.c, line ~580). This sets the attribute to a "known" value, and should fix the issue. For the second error code, see [this devzone post](https://devzone.nordicsemi.com/question/3376/svc-handler-missing/), and Ole Morten's answer. This is likely a toolchain/setup issue.

trond-snekvik, 2015-02-24:
@mrquincle , did this solve your issue, or is it really specific to the framework? If it worked out, I'll go ahead and close this one :)

mrquincle, 2015-02-24:
I solved this issue by making changes to your code. It is up to you to incorporate those changes or not. ;-) I didn't post this issue because I couldn't solve it. I did post it, so you are aware of the problem. The code is currently in our project - https://github.com/mrquincle/bluenet/tree/master/src/third/protocol - and I don't know if I have time to create a proper pull request.

Feel free to do with it whatever you like. :-)

trond-snekvik, 2015-02-24:
Ah, I see. I'll implement a fix for it in this repo as well, thanks:)

trond-snekvik, 2015-03-03:
I've gone a different route than you for this issue, but take a look at the fix in f17ff8a and see if it solves it.

daviddedwin, 2016-02-05:
Issue closed due to inactivity.

closed 2016-02-05 by daviddedwin