NordicPlayground/nRF51-ble-bcast-mesh #5: Any tips on getting this to compile with Keil 5.14 and SDK 7.2.0?

Zebble, 2015-02-21:
I keep getting conflicts with existing functions, and those that I've managed to get included through RTE Component Classes.

I must admit, I'm still a relative noob with Keil and the RTE/PowerPacks stuff.  I'm sure this is something simple.

Would really be great if there was a gcc method for compiling this library as well.
Thanks for the help!

trond-snekvik, 2015-02-25:
We're about to push a new release, which will improve upon the project-structure and SDK version support. SDK7.x.x will be comp

Zebble, 2015-02-27:
Great!  Looking forward to the update...

trond-snekvik, 2015-03-03:
The release was just pushed (v0.6.1), and includes UV5 project files with SDK7.x.x support (look for the _SDK7 suffix to project names. In addition, there's a section dedicated to deploying the folders locally in the correct manner in the readme, called "Getting started". GCC is still pending, but I will close this issue, as GCC-support is reported as a separate issue in Issue #4.

closed 2015-03-03 by trond-snekvik