NordicPlayground/nRF51-ble-bcast-mesh #6: Compilation help

kusterl, 2015-02-26:
Hi,
could you provide a short compilation instruction to get an example working on a nRF51, including what tools used and what versions?

I tried to compile it using a modified Makefile from an example of the nRF51_SDK_v6.1, which did not work because it somehow did not generate all object files. I also tried it with Keil uVision but I didn't manage to get my nRF51822 into the device database.

Thanks for any help

daviddedwin, 2015-03-06:
Is this a mesh specific issue, Can you get any of the SDK projects on SDK 6.x to compile ? If not you should ask this as a question in the Nordic devzone.

kusterl, 2015-03-09:
Now compilation works. The examples work without any problems so far. I had some problems with the project setup which I fixed in the meantime.
Btw. the new Readme is now much better on how to get it working.

closed 2015-03-09 by kusterl