NordicPlayground/nRF51-ble-bcast-mesh #10: adding rssi?

olskyy, 2015-03-24:
Hi there!

I'm trying to add an RSSI for each rbc_mesh_event. Can you please comment if am I doing it right?

1) extend "rbc_mesh_event_t" and "packet_t" with 
  > uint8_t rssi;
2) in the "search_callback", right after the checksum, I'm getting the rssi: 
  > uint8_t rssi = NRF_RADIO->RSSISAMPLE;
  and set on the packet:
  > async_evt.callback.packet.rssi = rssi;
3) so that rssi is available in the "mesh_srv_packet_process" with
  > uint8_t rssi = packet->rssi;
  and thus can be set in the event:
  > update_evt.rssi = rssi;
or conflicting event:
  > conflicting_evt.rssi = rssi;

Also, I have tried to config NRF_RADIO->SHORTS with RADIO_SHORTS_ADDRESS_RSSISTART_Msk, but it seems is having the same result as with just RADIO_SHORTS_READY_START_Msk. 

Can it be? Or should I just start it with: NRF_RADIO->TASKS_RSSISTART = 1; ?

Am I missing the point?

Thank you!
Cheers
Oleh

trond-snekvik, 2015-03-25:
Hi!

This all seems reasonable. You should start the RSSI with the short, like you mentioned, there is no need for the manual start task. You do, however need to make sure the RSSI capture has finished before you read it. Check the `NRF_RADIO->EVENTS_RSSIEND` register. Remember to flush it afterwards. 

You might have some trouble with the radio_controller hijacking your shorts, as it changes them a lot. I would recommend that you set up the RSSI-short from within that module, as to isolate all radio-manipulation. Just grep for writes to `NRF_RADIO->SHORTS` within radio_controller.c, and make sure that you always bitwise-or it with the RSSI-short. This should have no effect on the rest of the radio operation.

olskyy, 2015-04-13:
Hi! 

thank you for your reply.  Sometimes I do not get rssi value, so I thought to set a short delay to assure the capture. Obviously waiting too long is compromising other functionality. 
In the product specs I have found rssi-period of min. 8.8 us, setting the delay for 10us does not always bring the rssi-result. 
Now the question how tolerant will be the rbc-mesh when the delay introduced?

thank you!

olskyy, 2015-04-14:
just a short update...

after a bit more testing I've got the rssi with the delay of 10us, and it seems the values are reliable. 

thanks and cheers!

trond-snekvik, 2015-04-14:
That's good news :+1: 
For the record, the rbc-mesh _might_ have some trouble in some corner cases when you introduce a hardcoded delay into the system. It needs to make sure that the timeslot is ended before the Softdevice decides it's its own turn, and a delay might disturb the margin it has. I would recommend that you do enable the RSSI-interrupt, and treat it as a separate event in `void radio_event_handler(void)`, without any blocking. While a delay probably works as well, you might want to do some extensive testing to make sure that you don't overstep the Timeslot end.

AlexDM0, 2015-04-14:
Hi,
We've been playing with the RSSI as well and we also had the problem with missing values from time to time. It turned out we were not reading it at the right location. Our last tests have yielded successful RSSI values consistently. You could take a look in our code to see if our method also works for you.

https://github.com/dobots/bluenet/tree/master/src/third/protocol

You can search the code for rssi.

Regards,

Alex

olskyy, 2015-04-17:
Hi Alex, thanks for reply! 

you're right: placing the reading at the start of the function works all of the times ( mine was after CRC reading and around 8% of samples entered delay of 1-2 us. from the sample-test data of 10 days).

Now it never goes thru delay, thanks!

Cheers
Oleh

closed 2015-04-17 by olskyy

srd1, 2015-04-18:
Hey Alex and Trond, 

I am working on Home Automation (as Trond knows) and want to access the RSSI. Now the functions in end of radio_control.c which Alex had mentioned returns a uint8_t RSSI value. 

But what does that mean?

If I were a node in this mesh, I can see RSSI of my neighbours - right? If I have 4 neighbours, then whose value does this function return?

AlexDM0, 2015-04-18:
Hi,

As I understand it, and based on what we measured with a logic analyser, the moment a package comes in is very very brief. The measured RSSI value corresponds with the incoming package. If your package is readable you can assume that there is no crosstalk and your RSSI signal correspond to that sender. 

As Trond mentioned in another issue, the nodes do not talk 99.98% of the time. The time they send and a readable message was received, only one set of waves was sent into the radioband. That means there is no RSSI from nodes that are not sending. 

As for what it means in terms of distance, only testing can show this as it is a factor of your module type, environment and package. 

Regards

srd1, 2015-05-06:
Thanks Alex. That clarifies it.  

Can you highlight to me how can I actually access the name (variable or struct) of the node from which the message is arrived?

I can read the value, makes sense too, but tough to identify who does it belong to. So is there a variable that stores the name of the sender of the latest BLE message? Will be very helpful.

AlexDM0, 2015-05-06:
Thats an interesting point, @trond-snekvik, I don't think that is currently possible. If I'd want to insert an uuid address (say 4 bytes) to a packet without changing the version of the handle, I think I'll have to insert some code for that here correct? :

https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/transport_control.c#L225

and overwrite the last 4 bytes of my package for this?

Regards,

Alex

srd1, 2015-05-06:
I think there is a variable indeed, in struct ble_gap_addr_t in file ble_gap.h

evt->originator_address.addr

How about this snippet, Trond?

void rbc_mesh_event_handler(rbc_mesh_event_t* evt)
{
    switch (evt->event_type)
    {
        case RBC_MESH_EVENT_TYPE_CONFLICTING_VAL:	
        case RBC_MESH_EVENT_TYPE_NEW_VAL:	
        case RBC_MESH_EVENT_TYPE_UPDATE_VAL:	
      
			origin_name[0] = evt->originator_address.addr[0];
                        origin_name[1] = evt->originator_address.addr[1];
                        origin_name[2] = evt->originator_address.addr[2];
			origin_name[3] = evt->originator_address.addr[3];
                        origin_name[4] = evt->originator_address.addr[4];
                        origin_name[5] = evt->originator_address.addr[5];
			origin_type= evt->originator_address.addr_type;
                        
                        // catch RSSI value here. Or set a flag to catch it in main().
                        break;
    }
//... ....
}

srd1, 2015-05-06:
Are these originator addresses unique per node? Are these actually what I am looking for?

typedef struct
{
    rbc_mesh_event_type_t event_type;        /** See @ref rbc_mesh_event_type_t */
    rbc_mesh_value_handle_t value_handle;    /** Handle of the value the event is generated for */
    uint8_t* data;                      /** Current data array contained at the event handle location */
    uint8_t data_len;                   /** Length of data array */
    ble_gap_addr_t originator_address; /******** GAP address of node where this version of the message appeared ******/
		//uint8_t originator_rssi;			//srd: olskyy
} rbc_mesh_event_t;

typedef struct
{
  uint8_t addr_type;                    /**< See @ref BLE_GAP_ADDR_TYPES. */
  uint8_t addr[BLE_GAP_ADDR_LEN];       /*******< 48-bit address, LSB format. *******/
} ble_gap_addr_t;

trond-snekvik, 2015-05-06:
Hi,
The originator address supplied as part of the mesh packet (and eventually the rbc_mesh_event), is the address of the device that first introduced this value version into the mesh, and will therefore be the same each time the packet is retransmitted, regardless of which node does the retransmission. Further, packet conflicts are determined by [comparing the checksums of the packets](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/mesh_srv.c#L652), since comparing each byte in the payload is time consuming. If you start altering the packets (or the address included as part of the packets), the checksum changes, and the packets will be marked as conflicting. 

If you still want to change this, and get the RSSI and the address of the sender for each packet, this will require some alterations:
- First, make sure that the packets are transmitted with the local address, instead of the one you received along with the packet: Change the `md_ch->last_sender_addr` in [the memcpy in `mesh_srv_packet_assemble()`](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/mesh_srv.c#L748) to the local address, which can be pulled from the softdevice with `sd_ble_gap_address_get()`, as demonstrated on [mesh_srv.c, line 792](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/mesh_srv.c#L792). Now, each packet will contain the address of the actual sender, rather than the originator of the particular value version, without updating the version number in the packet.
- The second step is to drop [the crc-check mentioned earlier](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/mesh_srv.c#L652), and replace it with a memcmp or similar, where you check each byte of the packet data against the version we already had in our local database. Now we won't get any conflicting packets because of differences in the address field.
- The final step is to supply the RSSI along with the packet event, originating in the radio-controller. Activate the RSSI measurement with the radio SHORT, as mentioned in the beginning of this thread, and read the RSSI register at the end of the reception. I would recommend doing this [here](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/radio_control.c#L356), by changing the callback function signature to include RSSI, and call it like this: `(*prev_evt.callback.rx)(rx_data[!current_rx_buf], NRF_RADIO->RSSISAMPLE);`. This way, you can extract the address from the raw data (as is done already in [`packet_create_from_data()`](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/transport_control.c#L107), and send it to the upper layers along with the RSSI.

srd1, 2015-05-12:
Thanks Trond.

The line:
(*prev_evt.callback.rx)(rx_data[!current_rx_buf], NRF_RADIO->RSSISAMPLE);
refuses to compile as it does not admit 2 arguments!

trond-snekvik, 2015-05-12:
You have to redefine the function to let it take two arguments :)

AlexDM0, 2015-05-12:
@trond-snekvik would this be something you would consider putting in the mesh network? It seems a logical request. It allows localization, mapping etc.

Regards

olskyy, 2015-05-12:
+1 
I'd be happy to have this feature ootb. Currently I have just added an extra field to have the sender address, still having origin address.

trond-snekvik, 2015-05-13:
I'm finishing up my thesis over the next few weeks, so I don't think I'll have time just quite yet, but I'll see what I can do :)