NordicPlayground/nRF51-ble-bcast-mesh #13: TICK function

srd1, 2015-04-12:
Hey,

The mesh works fine. I wish to integrate it into my Home Automation related project. The hardware IC is nRF51822-QFAA-T. Software is Keil 5. 

In the BLE Gateway example, the rbc_mesh_event_handler() has first line as : TICK_PIN(28). 
Digging further, this TICK is used in many inside functions:

transport_control.c line 186: TICK_PIN(6);
rbc_mesh_common.h line 103: TICK_PIN(PIN_BIT_H); etc.

And every time it takes a "different pin number as argument".

The question is:
Are these TICKs for user debugging/LED toggling purpose or are they very crucial part of the drivers? 
Alternatively, if the code says TICK_PIN(6) and I want to use pin 6 for some sensor maybe, can I comment this TICK or should I prohibit myself from using pin 6 and use pin 5?

Awaiting reply, thank you. 
 
Srijan Dhare
Leaf Technologies Pvt Ltd

trond-snekvik, 2015-04-12:
Hi, the TICK_PIN() macro is indeed a debug macro, that simply enables a pin for ~0.5us, then disables it again. All usages can be disabled by setting the `RBC_MESH_DEBUG` flag in rbc_mesh_common.h to 0, or you may remove it completely. Either way, you can safely use the pins for other purposes afterwards. It has no real effect on the application, other than allowing for debugging with an external logic analyzer.

srd1, 2015-04-13:
Got it!

closed 2015-04-13 by trond-snekvik