NordicPlayground/nRF51-ble-bcast-mesh #14: Delayed Action

srd1, 2015-04-17:
Hey Trond, 

I put one node coded with program LED_mesh (in examples) in every room of my house of two floors. 
The node with BLE_Gateway (in examples) is kept in the Hall (First Floor) which connects to my phone by Nordic Master Control Panel App. 

Firstly, I had to keep a (extra) node at the stairs so that it is 'the one that connects' nodes at Ground Floor to the nodes at First Floor. Kind of acts as repeater. Firstly, can this be avoided? I know BLE range practically is quite suffocating. 

Secondly, when I toggle LEDs on the nodes by sending 0/1 by Nordic's App, the nodes at First Floor respond instantly, but the nodes on Ground Floor take (randomly) 1 seconds to 5 seconds to show the effect. The problem could be that the nodes at Ground Floor are connected to nodes at First floor by the repeater at stairs because if I remove that node then Ground Floor won't respond. So, can I reduce this delay of 1 to 5 seconds by changing a parameter or doing a tweak? I tried changing init_params.adv_int_ms to 10 (earlier it was 100), but doesn't make any difference, lag still there.

trond-snekvik, 2015-04-17:
Hi,
I'm afraid this issue is mostly related to signal quality decay over distance, causing you to lose packets at longer distances. The LED_mesh-nodes are in RX-mode in 99.98% of the time, and would receive anything that's on air. There are a couple of things you could do, though:
- Increase TX strength: In [`radio_init()`](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/radio_control.c#L388), change the line
 
`NRF_RADIO->TXPOWER      = ((RADIO_TXPOWER_TXPOWER_0dBm << RADIO_TXPOWER_TXPOWER_Pos) & RADIO_TXPOWER_TXPOWER_Msk);` 

to 

`NRF_RADIO->TXPOWER      = ((RADIO_TXPOWER_TXPOWER_Pos4dBm << RADIO_TXPOWER_TXPOWER_Pos) & RADIO_TXPOWER_TXPOWER_Msk);`

This boosts your transmissions somewhat, and your range should be extended. 

- Change the operating channel: This is done directly in the init-function in the API. Preferrably, use one of the advertising channels 37, 38 or 39, as they are not colliding with your Wifi-channel. The three channels are far apart in frequency, and you might be able to dodge some noise on your band by changing. If you have a lot of BLE devices lying around your house advertising, this might cause some disturbances as well. Try turning off those devices and see if you get better results.

- Like you said, you could reduce the adv-interval, but 10ms might be a little too short if you have several devices and trouble with packet reception. You risk saturation by having several devices babbling over each other, and it could make your problem worse. With 50ms+, the chance of getting collision creeps down below 1%, and you should be fairly safe. However, your devices don't have to operate on the same intervals, and it could be beneficial to have a short interval on the devices that appear to be bottlenecks (i.e. the staircase-device), and 50ms+ on the less important devices.

- Since you say that other devices react instantly, this is probably not your problem, but it could be worth mentioning: The Gateway node loses about 50% of its mesh-time because it has to keep up the connection to the Nordic-app. Unfortunately, the app doesn't have a setting to change the connection interval, you would have to use another similar application or the desktop-version of the Master control panel to fix this. Also, make sure that you use the LED_mesh example in all devices that don't require gateway functionality, especially for your stairs node, as it connects the two groups of nodes.

If none of these things work, I would recommend that you try to reduce the distance between the devices until you get satisfactory delays. If you need to go below 1-2 meters, there might be other issues at play.

srd1, 2015-04-18:
That is very elaborate, I got it. Safe would be to keep every node at 50+ ms. Also changed the dBm and noticed slightly better range. The channel I am using is 38. Turned off all other BLE devices in vicinity. This works fine!

closed 2015-05-04 by trond-snekvik