NordicPlayground/nRF51-ble-bcast-mesh #16: Cannot see messages from a node at Gateway after 2 minutes (if mesh has 5+ nodes)

srd1, 2015-04-18:
Hi Trond, 

Now there is a bigger problem. I have programmed all the nodes to update the packet every 30 seconds. Every node is given a unique 2-byte number as identity and whenever a node updates the packet, it also feeds these 2 bytes at index 2 and 3. The Gateway is calmly listening to all these packets and I am showing the "live feed" on my terminal. What I observed is, the nodes hang randomly. In an experiment with 8 nodes, as soon as I power them ON, each node's message can be seen on terminal. But slowly they start disappearing as if they are stuck (but I am sure they are not stuck, rather on debugging I found the application to rest inside arm_startup_nRF51.s at line 211). In other words, nodes work as soon as I reset them (soft or hard), but randomly their messages stop coming after some random 2-5 minutes and I am left with only 2-3 nodes still working out of 8. 

Did someone experience this? What could be the problem and solution, please?

srd1, 2015-04-18:
I think the issue named "Global time overflow" at 9 minutes is relevant. Should try that fix. 

What is your opinion Trond?

trond-snekvik, 2015-05-04:
Hi, sorry for the delay. I've pushed a fix for issue #7 in v0.6.3. Does the problem persist for this version?

closed 2015-07-27 by trond-snekvik