NordicPlayground/nRF51-ble-bcast-mesh #18: Mbed IDE porting

Pidjey, 2015-05-12:
Could this library be ported to Mbed ide? It will help me so mych.

Awesome project.

trond-snekvik, 2015-06-01:
Hi, sorry about the delay.

No Mbed port is planned for now, but it might be supported some time in the future.

codezork, 2015-07-14:
Hi,
I also would be interested in a portation to mbed. Keil is no solution for most of the potential users because they are not willing to pay  4000€ for a 1990 style IDE. The free version of Keil is useless as it is limited to 32K and the mesh solution needs by far more. 
What roughly speaking is neccessary to do for a portation (besides the upgrade to v9 of the nordic sdk) that this could be built under mbed? Are there other teams out there already planning this?
Cheers, Joerg

daviddedwin, 2015-07-15:
Almost all the Nordic examples will work well within the 32K limit of Keil. For example the mesh examples use about 13K to 15K of flash as seen from the Keil compiler. This is due to the fact that the softdevice (BTLE stack) is not linked in to the application i.e. the application can use the full 32K available.

So if you are evaluating Nordic you do not need to pay $$ to Keil or IAR, you get a full compiler and debugger toolchain for free using the size limited Keil version and the Nordic development kits that come with the Segger debugger.

This is clearly better than the online mbed IDE that does not allow debugging, so if you are seriously evaluating and doing product development , please consider a full IDE like Keil/IAR or Crossworks.

codezork, 2015-07-15:
I tried the proximity sample on Keil 5.15 and get 35140 bytes. Not yet using mesh. Is there anything wrong with what I am doing? 
[image]

daviddedwin, 2015-07-16:
I get this in my keil 4.72a build using ble_app_proximity_s110_pca10028 in SDK 8.1

Optimization level 3:
Program Size: Code=16408 RO-data=344 RW-data=208 ZI-data=3712 
Approximately 17K of Flash.

Optimization level 0:
Program Size: Code=24996 RO-data=348 RW-data=228 ZI-data=3708  
Approximately 25K of Flash.

==
Keil 5.14 : Zip files : No packs used : ble_app_proximity in SDK 8.1

Optimization level 3:
 Program Size: Code=16416 RO-data=344 RW-data=208 ZI-data=3712 
Approximately 17K

Optimization level 0:
 Program Size: Code=25088 RO-data=348 RW-data=232 ZI-data=3712 
Approximately 25K of Flash.

I am using the zip files ( I do not use packs on Keil 5).
=

As you can see the size of the project is well below 32K.
You are using the wrong project, I think you should be using the ble_app_proximity_S110_pca10028 and not the serialized versions (hci/spi etc).

codezork, 2015-07-16:
Thank you very much. This worked. I had too much modules linked I think. I don't understand right now what makes the difference between pack and zip.

trond-snekvik, 2015-08-19:
There is now an option for building with GCC included in the framework. While it's not mbed, it might be useful for others reading this because they don't want to use keil

daviddedwin, 2016-01-27:
mbed is not in our immediate roadmap.

closed 2016-01-27 by daviddedwin