NordicPlayground/nRF51-ble-bcast-mesh #19: Restriction of Nodes number in Mesh

Carigll, 2015-05-13:
Hi Trond,

I've tested with ten nodes, and it runs quite all right though. While I've check the illstration docs, and code too. I see there is no mention about the restriction of node's number.So does it have a limination that how many nodes could be put into this mesh?

BRs.

trond-snekvik, 2015-05-13:
Hi,
there is actually no restriction on the number of nodes in the system! Since they don't address each other (or keep track of their neighbors), the devices don't need to store ID's or lists of neighbors, and the amount of devices makes no difference.

While we don't encourage it, this means that you could theoretically establish a global network of a billion evil minions that communicate through this mesh solution to secure your world domination. We haven't listed this as a potential usage scenario though, since the value propagation time increases with area, and you might get a lot of conflicting value updates when all your minions start blabbering over each other.

closed 2015-05-13 by trond-snekvik

olskyy, 2015-05-13:
I love the idea of nRF-Minions network :-D :+1: 
@trond-snekvik : you rock! :100:

trond-snekvik, 2015-05-13:
Quick, trademark the term nRF-Minions! Whole new market segment, right there