NordicPlayground/nRF51-ble-bcast-mesh #22: Help regarding use of Timeslot API

strawhat5, 2015-05-25:
I am building a project in which I need to establish peer-to-peer communication between 2 nRF51422. I came across a post which said that I can achieve that using Timeslot API. If so, how can I can use Timeslot API to setup the connection?
Or, can this project "nRF51-ble-bcast-mesh" be used directly to setup peer-to-peer communication? If so, please guide me on how to use it to setup peer-to-peer communication.

trond-snekvik, 2015-06-01:
Hi, this project does what you want, but with some restrictions. The power consumption is high, and you are only allowed to flood messages to all devices in the network. The mesh-solution attempts to synchronize a set of handle-value pairs across all devices in the system. A handle-value pair could represent various states in the system, like "Living room lights on/off" or "Refridgerator Temperature". 

As a start, I would recommend that you run the LED_mesh example under /nRF51/Examples/, and get familiar with the api presented in /nRF51/rbc_mesh/rbc_mesh.h. Read the README, and accompanying files under the /docs/ folder. As a basis for your own application, use the template project at /nRF51/examples/Template project, and build from there.

strawhat5, 2015-06-01:
Thanks for your guidance. I have started using this project since a while ago but I am stuck at one point.
I am currently experimenting on the BLE-Gateway-SDK7 example. I wanted to send my "Custom Data", so I inserted the "Manufacturer Data" in nrf_adv_conn.c. And, I flashed the "edited" BLE-Gateway-SDK7 on to my 1st nRF51422. Now, I wanted to read this Custom Data using my 2nd nRF51422 and make changes to it. So, how should I achieve this? I tried flashing the "unedited" BLE-Gateway-SDK7 on to my 2nd nRF51422, but I didn't know how to proceed further. Or, should I need to change something else in the BLE-Gateway-SDK7 project?

trond-snekvik, 2015-08-07:
I hadn't seen your reply, sorry.
I would recommend that you read the [structure document](../blob/master/docs/structure.adoc), which explains how you should interact with the framework. All nodes share the same data set, and will attempt to synchronize all variables across the mesh. They're also all allowed to read or update all of them through the API. When a node updates a value with `rbc_mesh_value_set()`, the rest of the nodes get an event from the framework, notifying them of the change. They can choose to act on these events (by turning on leds for instance) or just let them be, and store them in their local copy of the shared data set.

closed 2016-02-05 by daviddedwin