NordicPlayground/nRF51-ble-bcast-mesh #23: reconnected failed

eroswade, 2015-06-04:
while i fellow the introduce 'integrating_w_SD_apps.adoc' to set the example 'ble_app_proximity' . 
it works fine with softdevice s110_nrf51822_7.0.0, 
when i work with 'experimental_ble_app_uart'  it works fine either with 'master control panel' connect.
but if i disconnect, then it can't be scan anymore. 
then i found if BLE_GAP_EVT_TIMEOUT occur and using advertising_start to start adv. it's succeed. but something wrong in 'sys_evt_dispatch'
when i pause the project. it runs on 
HardFault_Handler\
                PROC
                EXPORT  HardFault_Handler         [WEAK]
                B       .
                ENDP
i upload the proj : https://github.com/eroswade/NordicPrist.git
anyone can help 

thanks

trond-snekvik, 2015-06-04:
Hi,

When you disconnect from the master control panel, the nRF51 should give you a BLE_GAP_EVT_DISCONNECTED if it detected the disconnection. Check this one as well.

I'm afraid I don't understand what you mean by pausing the application, could you explain? Do you stop it from Keil or set a breakpoint?

The hardfault handler is called when you access a peripheral illegally, for example if you use the radio outside a timeslot.

eroswade, 2015-06-04:
thank you for your replay. 
yes i reviced the BLE_GAP_EVT_DISCONNECTED, and using advertising_start() to restart the advertising. or even i set time out, and i received BLE_GAP_EVT_TIMEOUT, also i called advertising_start(). 
this two method will cause the same problem . the advertising is stoped. and i click keil with "STOP", it stoped in arm_startup_nrf51.s with line HardFault_Handler.

eroswade, 2015-06-05:
I've finished the test 
and thank for Rallare , who send a pull request for you. 
diff
+ 
+Due to restrictions in the timeslot API, you will need to set the advertising interval in your application (normally define "APP_ADV_INTERVAL_FAST") equal or greater to the timeslot period used by rbc-mesh library. The timeslot period is now set to 100 ms, which means that the fastest advertising in your application shall be >= 100 ms.

closed 2015-06-05 by eroswade