NordicPlayground/nRF51-ble-bcast-mesh #24: Is there a product anomaly preventing old version to work with mesh?

mrquincle, 2015-06-04:
We have been using the meshing functionality for a while on a diversity of chips, also the `QFAA` packet/variant. The evaluation kit has hardware code `003C` and works fine. However, we have also older chips with hardware code `001D`, but the same packet/variant, and meshing will in this case not work parallel to a connection request.

The meshing works fine, even parallel to the device in its normal broadcasting state for BLE. However, as soon as I try to actually establish a connection, there is some fault within the SoftDevice (S110).

```
Program received signal SIGTRAP, Trace/breakpoint trap.
Read 4 bytes @ address 0xFFFFFFFE (Data = 0x00000002)
0xfffffffe in ?? ()
(gdb) bt
#0  0xfffffffe in ?? ()
#1  <signal handler called>
Read 4 bytes @ address 0x20003F74 (Data = 0x21000011)
Reading 64 bytes @ address 0x20003F40
Read 4 bytes @ address 0x000000FE (Data = 0x47085809)
```

Personally, I think this might be related to some kind of product anomaly for this older chip, but do you have some idea about what it could be? It only happens on a connection request and it is very consistent. I hope you have an older chip like this somewhere, because it is then super easy to reproduce.

Of course, feel also free to test our code at https://github.com/dobots/bluenet if that is of any help in testing this scenario.

mrquincle, 2015-06-04:
One thing I found is PAN 20, which states the RADIO->STATE register is not functional for the CA/Cx0 (`001D`) build codes (contrary to the `Gx0` build code for the `003C`). However, it doesn't look like you actually use that register, so it must be something else.

daviddedwin, 2015-07-15:
Is the issue still open ? 
If this is still an issue, please mention the Softdevice versions that you are using.

mrquincle, 2015-07-15:
Yes.

S110, newest version.

daviddedwin, 2016-02-05:
We recommend that all work be moved to the rev 3, nRF51 ICs (sdk-8-support).  Rev 1 nRF51 ICs will not supported.

closed 2016-02-05 by daviddedwin