NordicPlayground/nRF51-ble-bcast-mesh #32: Build mesh code with GCC toolchain

foldedtoad, 2015-07-25:
I have ported the code over to the GCC toolchain (https://launchpad.net/gcc-arm-embedded).
This a CLI build method, not from an IDE.
There were several changes which I made, but nothing too drastic.

I would like to contribute these changes.
I would think that allowing me to create new  branch, say "sdks-8-support-gcc", and allowing me to check my code into this branch would be one way.
Alternatively, I could create a project on my github account and check my changes in there.
But this would split the code base.

Can you tell me how to proceed?

trond-snekvik, 2015-07-27:
That's awesome! :smiley: 
We'd love to get your port into the repo, we've been putting this task off for months.

I think the best way would be if you were able to put in a pull request. I think Github requires that the pull request is from a repo that has been forked from the original, but I think this is the cleanest way. I think this belongs on the master branch, as a separate branch would just lead into meaningless additional work for each new version of the mesh.

Does this work for you?

foldedtoad, 2015-07-28:
Just create a pull-request.
I am sure there will be questions, so feel free to ask.
See the comments on the pull-request for details.

trond-snekvik, 2015-08-19:
I chose to include your contributions from under the /gcc/ folders. While the util would be useful in future releases, we don't want to add it just yet. We'll credit you if we add it in later:)

Thanks for giving us a much needed extra push for this feature, it works perfectly :+1:

closed 2015-08-19 by trond-snekvik

dburongarcia, 2015-09-30:
Hi, I know this issue is closed, but I would like to know where is the code for GCC and which SDK uses (is it 8.0.0?) I couldn't compile the examples with gcc on OS X Yosemite, any tips? I followed the setup of electronut http://electronut.in/nrf51-adc-test/ and everything works there. Thanks :)

trond-snekvik, 2015-09-30:
Hi, GCC makefiles are available in all the example projects under _~/nRF51/examples/\<EXAMPLE NAME\>/gcc/_ in the sdk-8-support branch. The master branch, which is targeted at SDK7 and lower is currently parked, as we are in the process of moving to only support SDK8.1+. 

The GCC (and ARMCC) builds require SDK8.1 to compile. It is important that you clone (or unzip) the repository in the _/examples/_ folder of the SDK to get all path references right. Also make sure your _/components/toolchain/gcc/Makefile.\<PLATFORM\>_ file is configured correctly. Let us know if there are any further problems.

dburongarcia, 2015-09-30:
Thanks, I'll try as you say and let you know.