NordicPlayground/nRF51-ble-bcast-mesh #36: Cannot compile examples with ARM-GCC

rifatmahmud, 2015-08-28:
When I use make on the example folder, it says target not found.
Has anyone been able to compile the examples in arm-gcc?

trond-snekvik, 2015-08-28:
Hi, the gcc-functionality has been tested with both Ubuntu 13.04 w/ arm-none-eabi-gcc v4.8.3, as well as Win7x64 w/Cygwin x86 1.7.32 & arm-none-eabi-gcc 4.8.4, and builds on both without modifications, so this is most likely a config issue.

Checklist:
What toolchain & OS are you on? 
Are you able to build any of the nRF51 SDK8.1 examples?
Where have you cloned the mesh-repo?
Have you entered the correct information in the _<SDK8-dir>/components/toolchain/gcc/Makefile.[system-name]_ ? For linux, use the _/usr/bin/_ version of the arm gcc.
Could you do a `make clean`, then `make` from _<project-dir>/nRF51/examples/BLE_Gateway/gcc_, and paste the full output?

We'll figure this out, but I need something more to go on:)

rifatmahmud, 2015-08-28:
I am using Ubuntu 14.04 64 bit.
Only when using pre-built arm-none-eabi-gcc v4.8.1 extracted to /usr/local I am able to compile nrf SDK examples and with that, I am also able to compile <i>BLE_Gateway</i> example. Installing from launchpad repository is not able to compile the SDK examples orany of your example.

LED_Mesh is throwing an error even with installation on /usr/local.

Should I open another issue for that?

trond-snekvik, 2015-08-28:
I think we had some similar issues with the launchpad version as well, but if the pre-built version works for you, we'll put it on the backlog.

The LED_Mesh issue could be related, what's the compiler output for that one?

rifatmahmud, 2015-08-31:
LED_Mesh compiler output on running 'make:

'Compiling file: mesh_aci.c
../../../rbc_mesh/src/mesh_aci.c: In function 'serial_command_handler':
../../../rbc_mesh/src/mesh_aci.c:222:57: error: too many arguments to function 'rbc_mesh_value_get'
                                                         &origin_addr);
                                                         ^
In file included from ../../../rbc_mesh/include/mesh_aci.h:38:0,
                 from ../../../rbc_mesh/src/mesh_aci.c:35:
../../../rbc_mesh/rbc_mesh.h:231:10: note: declared here
 uint32_t rbc_mesh_value_get(uint8_t handle,
          ^
make: *** [_build/mesh_aci.o] Error 1

rifatmahmud, 2015-08-31:
The whole repo is copied inside ~/BLE_Mesh

rifatmahmud, 2015-08-31:
Scaling_Example error:

Compiling file: nrf_drv_gpiote.c
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:30:61: error: '                                                  GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS' undeclared here (not in a function)
     nrf_drv_gpiote_evt_handler_t handlers[NUMBER_OF_GPIO_TE+GPIOTE_CONFIG_NUM_O                                                  F_LOW_POWER_EVENTS];
                                                             ^
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c: In function 'n                                                  rf_drv_gpiote_init':
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:148:44: error:                                                   'GPIOTE_CONFIG_IRQ_PRIORITY' undeclared (first use in this function)
     nrf_drv_common_irq_enable(GPIOTE_IRQn, GPIOTE_CONFIG_IRQ_PRIORITY);
                                            ^
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:148:44: note: e                                                  ach undeclared identifier is reported only once for each function it appears in
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c: In function 'p                                                  in_from_port_get':
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:93:1: error: co                                                  ntrol reaches end of non-void function [-Werror=return-type]
 }
 ^
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c: In function 'c                                                  hannel_handler_get':
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:98:1: error: co                                                  ntrol reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
make: *** [_build/nrf_drv_gpiote.o] Error 1
ubuntu@ip-172-31-12-65:~/BLE_Mesh/nRF51-ble-bcast-mesh/nRF51/examples/Scaling_example/gcc$ make                                   Compiling file: nrf_drv_gpiote.c
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:30:61: error: 'GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS' undeclared here (not in a function)
     nrf_drv_gpiote_evt_handler_t handlers[NUMBER_OF_GPIO_TE+GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS];
                                                             ^
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c: In function 'nrf_drv_gpiote_init':
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:148:44: error: 'GPIOTE_CONFIG_IRQ_PRIORITY' undeclared (first use in this function)
     nrf_drv_common_irq_enable(GPIOTE_IRQn, GPIOTE_CONFIG_IRQ_PRIORITY);
                                            ^
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:148:44: note: each undeclared identifier is reported only once for each function it appears in
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c: In function 'pin_from_port_get':
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:93:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c: In function 'channel_handler_get':
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:98:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
make: *** [_build/nrf_drv_gpiote.o] Error 1

trond-snekvik, 2015-08-31:
I'm afraid the first one is caused by a slight oversight which will be fixed for the next release. The `rbc_mesh_value_get()` function has been stripped of the "origin_addr" parameter as it is no longer relevant (see [release log for v0.6.8](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/releases/tag/v0.6.8-SD8)), but the serial-code fell through the cracks when doing the release. To fix it, simply remove the origin address parameter from the function call. 

The second one is a bit stranger. It appears as if you have a mixture of SDK8.0 and 8.1 (or 9.0?) installed. The `nrf_drv_gpiote.c` file was introduced in SDK8.1, but you seem to have some header files still on 8.0. The `GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS` constant is defined in _../../../../../../components/drivers_nrf/config/nrf_drv_config.h_ in SDK8.1, but not at all in SDK8.0. I would recommend that you download SDK8.1, and wipe any remaining SDK8.0-files. Since the config file was present (but different) in SDK8.0, you might have ended up not overwriting it when copying over SDK8.1 (?)

Also, I recommend cloning the mesh example inside the _/examples/_ folder of the SDK, to avoid any config issues.

closed 2015-09-16 by rifatmahmud