NordicPlayground/nRF51-ble-bcast-mesh #39: BLE Gateway example, impossible to write more than 16Byte to Gatt-Characteristic whit SD-8

vladi4k, 2015-09-10:
Hi, if I write more than 16 Byte on Mesh-Characteristic I get return code 12 <=> <b>NRF_ERROR_DATA_SIZE</b> in <b>[sd_ble_evt_get()](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/examples/BLE_Gateway/main.c#L111-116)</b> 
by <b>[BLE Gateway example](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/tree/sdk-8-support/nRF51/examples/BLE_Gateway)</b> -> main.c -> [sd_evt_handler()](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/examples/BLE_Gateway/main.c#L111-116) : Line 111

Additional I can't more connect to Device again

https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/examples/BLE_Gateway/main.c#L111-116

```c
/**
* @brief Softdevice event handler 
*/
uint32_t sd_evt_handler(void)
{
    rbc_mesh_sd_irq_handler();
    
    ble_evt_t ble_evt;
    uint16_t len = sizeof(ble_evt);
    while (sd_ble_evt_get((uint8_t*) &ble_evt, &len) == NRF_SUCCESS)
    {
        nrf_adv_conn_evt_handler(&ble_evt);
    }
    return NRF_SUCCESS;
}

Eclipse-GCC-ubuntu14-64Bit-SD8-Nexus5-nRFMCP-PCA10028

trond-snekvik, 2015-09-11:
You are correct, that is a slight miscalculation on my part. I'll put it up for the next release, but in the meantime, here's the fix: 
```
uint32_t sd_evt_handler(void)
{
    rbc_mesh_sd_irq_handler();
    
    uint8_t ble_evt[sizeof(ble_evt_t) + RBC_MESH_VALUE_MAX_LEN];
    uint16_t len = sizeof(ble_evt);
    while (sd_ble_evt_get((uint8_t*) &ble_evt, &len) == NRF_SUCCESS)
    {
        nrf_adv_conn_evt_handler((ble_evt_t*) &ble_evt);
    }
    return NRF_SUCCESS;
} 
```

vladi4k, 2015-09-11:
Hi, thank you for your help, but now it is 20Byte max only.

trond-snekvik, 2015-09-14:
Ah, yes. This is a Softdevice (and BLE spec) limitation; if you want more than 20 bytes write command length, you need to do a "long write". This is fully possible with the mesh, but is not done in the example, as it involves a bit of additional clutter. For more information, see [this devzone post](https://devzone.nordicsemi.com/question/155/writing-more-than-20-bytes-at-a-time-to-a-characteristic/) and [this github example](https://github.com/NordicSemiconductor/ble_app_hrs---LongWrite). 

If you have any additional issues with long writes, I recommend that you post a question on the Nordic devzone, as the Nordic tech support team is more likely to pick it up and give you some proper assistance :)

rifatmahmud, 2016-01-06:
https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/commit/8b3c55b3d40c65e96334d187816eb55aeb014382#commitcomment-15284031

closed 2016-02-05 by daviddedwin