NordicPlayground/nRF51-ble-bcast-mesh #40: Not working with s130 softdevice

itrs, 2015-09-11:
Using SDK 9, S130, and Gateway example with GCC.

assert failed at <Mesh Gateway>/nrf_adv_conn.c:167 with error_code 12

165│     /* Set advertisement data with ble_advdata-lib */
166│     error_code = ble_advdata_set(&ble_adv_data, NULL);
167├>    APP_ERROR_CHECK(error_code);
168│

trace and check value of ble_adv_data, it's mangled somewhere else so the content of ble_adv_data is not valid.

Maybe potential overflow somewhere else ?

BTW some code works fine with S110 configuration.

trond-snekvik, 2015-09-14:
Hi, error code 12 is NRF_ERROR_DATA_SIZE, and in this case, this means that the advertisement length exceeds the advertisement 31 byte maximum. I don't have the numbers in front of me, but I assume that the S130 attempts to add more auxiliary data to the advertisement than the S110 does. Does removing (or shortening) the name of the device fix your problem? Are you able to see where in the `ble_advdata_set()` procedure (or its subprocedures) the error originates?

itrs, 2015-09-16:
no, this is not cause of the issue. the issue is the ble_adv_data is totally corrupted somewhere. (checked content and there are meaningless fields). I can't find any control path doing this so I assume something like memory overflow.

Is there anyone have tried using bcast-mesh with S130 stack ?
Any example are welcome.

trond-snekvik, 2015-09-17:
Could you post the resulting ble_adv_data as it is when you get the crash?

VonRehberg, 2015-11-26:
Hey guys,

I wondern how you were able to utilize rbc_mesh with softdevice s130 at all!?
When trying to run the sample code on s110 everything works fine. But whenever I flash softdevice s130 and switch all code references accordingly, nothing gets executed and I'm unable to debug.
Am I missing something?

Thanks in advance,
Joschi

trond-snekvik, 2015-11-27:
The S130 v1.0 has been tested with the SDK8-branch, and should work if configured correctly. Did you change the scatter-file (for gcc) or the target-address dialog in Keil? The S130 is differently sized than the S110, and the application start address must be different.

VonRehberg, 2015-11-27:
Thanks for your fast reply.
Yes I changed it in Keil, but I'm not sure whether it is correct as I always have trouble finding the correct addresses (I'm new to this ;-) )
Currently I'm using IROM start address 0x1C000 and size 0x24000
For IRAM I'm using 0x20002800 and size 0x5800.
Is that correct?

trond-snekvik, 2015-11-27:
I think your IROM address is off. The [S130 Specification (pdf-warning!)](http://www.nordicsemi.com/eng/nordic/download_resource/33863/4/28090693), chapter 10.1 gives you the correct offsets. Try 0x20000 instead of 0x1C000, and size 0x20000. 
I think your RAM-offset is alright though.

VonRehberg, 2015-11-27:
Thanks for your help!
Unfortunately nothing has changed. Maybe I messed around to much.
I think I will start over from the examples this evening and try to switch from S110 to S130.

trond-snekvik, 2015-11-27:
Hm, maybe. If that doesn't help, try starting the program in debug mode in Keil, and set breakpoints in your app_error_handler() and HardfaultHandler(), and see if they're hit. Once initialized, the framework is very sensitive to timing, and will crash if you try to execute after hitting breakpoints or any code stepping, so be careful. Report back with any errors, and I'll see what I can do.

daviddedwin, 2016-02-05:
The S130 v1 and v2 and the bcast mesh should work together. Start with an S130 peripheral project and then add the mesh to it.

closed 2016-02-05 by daviddedwin