NordicPlayground/nRF51-ble-bcast-mesh #43: What's suggested schema to control single node of mesh ?

itrs, 2015-09-17:
Currently whole mesh share a view of handle-value pair table. What's suggested way to send message to specified node ? My idea is to specify each node with a id and associate the id to handle. So handle 0 is for node 0, handle 1 is for node 1 ... .Any other suggestion on such purpose schema design ?

trond-snekvik, 2015-09-17:
This is the way we currently recommend, but it has some issues - if two devices in different parts of the mesh try to write to node 1 at the same time, they will do so with the same version number, but different data. This triggers a conflict in devices that lead up to the target device, and this device has to choose which packet to forward. 

An alternative approach which doesn't have this problem, is to switch it around, so that each handle is only seeded by a single device. For unicasting, you could add your own address field at the beginning of the payload, but this would obviously alter your effective payload length.

Choosing between these two is obviously a tradeoff, and you should use whatever fits your application :)

olsky, 2016-02-18:
I was facing the same dilemma, first tried to allocate a handle per node, but it was not very effective in my case. For the test case I was using some of older rev chips with less memory, so the number of handles was limited, thus with more nodes in test -- the handles will become a bottleneck.

Currently I'm experimenting with round-robin scheduling, cycling thru handles, so each node keeps a track of last used handle and will use the next one for transmission. When the last handle is reached -- it will drop the the first available. 

I have also assigned one handle to keep track of last used handle, so it helps to synchronize it a bit :-)

trond-snekvik, 2016-02-19:
@olsky The sdk-8-support branch has a built in LRU-cache for handles, have you experimented with the settings for it? It allows you to control the memory footprint on each device, so that you don't have to scale with the number of devices.

olsky, 2016-02-19:
@trond-snekvik thank you, I will check it.

do you mean changing the RBC_MESH_HANDLE_CACHE_ENTRIES and RBC_MESH_DATA_CACHE_ENTRIES values?

trond-snekvik, 2016-02-19:
Yes :)

olsky, 2016-02-19:
that was fast! :-) thanks!

closed 2016-12-01 by trond-snekvik