NordicPlayground/nRF51-ble-bcast-mesh #45: Cannot compile Scaling_example in arm-gcc.

rifatmahmud, 2015-09-29:
I am using Ubuntu 12.04.02 LTS(64 bit).
I have cloned the whole repo inside nRF51 SDK's examples folder.

This is the error I am getting when I am running make on Scaling_example:

Compiling file: nrf_drv_gpiote.c
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:30:61: error: 'GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS' undeclared here (not in a function)
     nrf_drv_gpiote_evt_handler_t handlers[NUMBER_OF_GPIO_TE+GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS];
                                                             ^
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c: In function 'nrf_drv_gpiote_init':
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:148:44: error: 'GPIOTE_CONFIG_IRQ_PRIORITY' undeclared (first use in this function)
     nrf_drv_common_irq_enable(GPIOTE_IRQn, GPIOTE_CONFIG_IRQ_PRIORITY);
                                            ^
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:148:44: note: each undeclared identifier is reported only once for each function it appears in
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c: In function 'pin_from_port_get':
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:93:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c: In function 'channel_handler_get':
../../../../../../components/drivers_nrf/gpiote/nrf_drv_gpiote.c:98:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
make: *** [_build/nrf_drv_gpiote.o] Error 1

trond-snekvik, 2015-09-29:
Hi, this seems to be the [same issue as you had before](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/issues/36) (?)

Did you try my suggestion?

> It appears as if you have a mixture of SDK8.0 and 8.1 (or 9.0?) installed. The `nrf_drv_gpiote.c` file was introduced in SDK8.1, but you seem to have some header files still on 8.0. The `GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS` constant is defined in _../../../../../../components/drivers_nrf/config/nrf_drv_config.h_ in SDK8.1, but not at all in SDK8.0. I would recommend that you download SDK8.1, and wipe any remaining SDK8.0-files. Since the config file was present (but different) in SDK8.0, you might have ended up not overwriting it when copying over SDK8.1 (?)

Only the _nrf_drv_config.c_ file associated with 8.1 should have the offending code line, but only the 8.0 _nrf_drv_config.h_ file should miss the #define.

rifatmahmud, 2015-09-29:
It was a fresh Ubuntu install, and, I have unpacked SDK-8.1.0 from here:
http://developer.nordicsemi.com/nRF51_SDK/nRF51_SDK_v8.x.x/

So, I am uncertain how it could be a conflict between 8.1.0 and 8.0.0. I used your hotfix-issue-44 branch. The nRF SDK is extracted in NRF_SDK folder in root home folder, and the ble-bcast repo is cloned inside NRF_SDK/examples folder.(Is that a correct procedure? Or do I need to put the 3 folders inside https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/tree/hotfix-issue-44/nRF51
on the nRF51 SDK's examples folder?

rifatmahmud, 2015-09-29:
[image]

Here is the snapshot of my ...drivers_nrf/config/nrf_drv_config.h file.
The problem seems to be GPIOTE_ENABLED 0. When I defined it as 1, the Scaling Example compiled. 

What is the significance of this, and how this change the SDK's headers might effect other examples.

Thanks.

trond-snekvik, 2015-09-29:
You are correct. I have to admit, I didn't consider the flag. The config file in _/components/_ is meant to be used as a template for a project specific version, but in this case, we were a little too fast, and used it directly. We'll put a customized header in the example folder.

closed 2015-10-15 by trond-snekvik