NordicPlayground/nRF51-ble-bcast-mesh #46: Using examples in NRF Beacon Kit

dburongarcia, 2015-10-14:
Hi, I've been testing some code in both the dongle PCA10031 and NRF51DK PCA10028 using gcc and OS X and everything is working great. Wonderful work! :+1: 

But I'm having some trouble trying to update the NRF Beacon PCA20006 Rev 1.1 with the code, even with the examples, both gateway and LED mesh. Any tips? I've been trying the DFU of nRF Master Control Panel in Android because the NRFGoStudio is not working in OS X.

(I've bee also trying to get NRFGoStudio work both through playonMac, not installing, and Windows 7 with VirtualBox, but I'm having difficulties as it's not detecting the Segger programmer of the NRF51DK and a BT4.0 dongle for DFU neither) Thanks

trond-snekvik, 2015-10-14:
Hi, I think you would be better off trying to use the Segger somehow. The mesh-framework and examples aren't made with DFU in mind, and I'm not sure if a bootloader mode is even possible right out of the box. The Segger should however be able to work just as easily as for the DK's, and the nrfjprog tool alone should be possible, if you can get the drivers up and running. I suggest asking the [devzone](devzone.nordicsemi.com) or try the [support portal](https://www.nordicsemi.com/eng/supportcase/create) (you have to register as a user), as they are far better prepared to help you with these issues :)

dburongarcia, 2015-10-15:
In theory, with the rev 1.1 beacons DFU programming of any kind would be possible, but it asks me for the .hex and .dat files. I'm using OS X so nrfjprog it's not an option, I'm using JlinkExe.
In case I could program the beacon with the Segger, do I need to change anything form the example code to work in the beacons? It seems it just support the development kits and dongles. The beacons just have a RGB LED. Thanks, I will follow your tip.

vidarbe, 2015-10-16:
Jlink commander has the same functionality as nrfjprog/nRFgo studio. However, as you may be know, the beacon kit comes with 16K of RAM as opposed to 32K on PCA10031 and PCA10028 so the linker settings must be adjusted accordingly. 

With regards to programming, it is possible to use the DK kit to program the beacon kit as explained in this blog post: https://devzone.nordicsemi.com/blogs/686/programming-the-nrf51822-beacon-kit-with-the-nrf51/. I guess DFU should work as  long as you have the correct softdevice version, correctly linked application, and that there is enough space in the application region, but haven't tried it with this project.

trond-snekvik, 2015-10-21:
Update on this: there is now a [Linux-version of nrfTools (including nrfjprog)](http://www.nordicsemi.com/eng/nordic/Products/nRF51822/nRF5x-Tools-Linux/51386), could it be of any help?

closed 2016-02-05 by daviddedwin