NordicPlayground/nRF51-ble-bcast-mesh #47: OriginAddress in SD8?

olsky, 2015-11-04:
Hi Trond,

I notice the following API has been changed with no origin_addr in the new version:

uint32_t rbc_mesh_value_get(rbc_mesh_value_handle_t handle,
    uint8_t* data,
    uint16_t* len,
    ble_gap_addr_t* origin_addr);

could you please share the reason? If it is just old documentation, can I get still the last sender address?

Thank you!
Oleh

trond-snekvik, 2015-11-05:
Hi, 
You're right, that documentation missed the update.
The origin address was removed when we switched over to putting the actual local adv address in the packet (as per BLE spec) instead of the data-originator address. As a single message can come from any device in the network without that device necessarily having any relationship to the packet, the adv-address can't really help you making any mesh-related decisions, and we decided to remove it to avoid confusion. The sender address is registered internally, but discarded by the transport layer before it's propagated, so it should be possible to fetch it again if you're willing to get your hands dirty :)

olsky, 2015-11-12:
Thanks, it makes sense :)

closed 2015-11-12 by olsky