NordicPlayground/nRF51-ble-bcast-mesh #49: Custom payload support in rbc sd8?

olsky, 2015-11-04:
Hi Trond,

as you moved to the spec-conformat pachet format, it opens new possibilities; like transmitting and receiving custom payloads. I would like to experiment with it a bit. What would be the best place in the rbc to place such a listener of custom payload? 

Thank you!

Cheers
Oleh

trond-snekvik, 2015-11-05:
The best way might be to do it in the transport control-function `tc_packet_handler()`. The mesh-specific code starts at [line 228](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/src/transport_control.c#L228), and can be isolated from the rest. I would L228-L294 in a separate function (call it mesh_packet_handler or something), then handle generic packets in a similar function either right before or right after.

trond-snekvik, 2015-11-05:
As for TX, you can use `tc_tx()` directly - it doesn't care about packet format. Just remember to use the packet-module for allocation/reference counting.

olsky, 2015-11-09:
Awesome! thanks, now its time to code a bit :)

closed 2016-02-05 by daviddedwin