NordicPlayground/nRF51-ble-bcast-mesh #51: SD-less bcast version

olsky, 2015-11-12:
Hi @trond-snekvik 

I was going thru the source code from your last release which I do like a lot personally and find it very modular and clean. Especially no gatt binding. But with such a great separation there is no need in SD at all... :-)

There could be a build or version with the pure NRF_RADIO usage. Of course when no SD functionality is used. This would make sense for nodes playing the rely role.

What are your thoughts on this?

Cheers
Oleh

trond-snekvik, 2015-11-12:
It has crossed my mind, and you can actually see traces of it in the code (notice how mesh_gatt is [encapsulated with a `#ifdef SOFTDEVICE_PRESENT`](../blob/sdk-8-support/nRF51/rbc_mesh/src/mesh_gatt.c#L37)? ;) ).

Currently, we don't want to expose the possibility in the project through examples or similar, as we think this dramatic configuration split will increase the support-cost for the project, and introduce some issues we haven't considered before. 

For now, we intend to leave this to the experienced/heavily invested application programmers, but an SD-less example might be an option in the future. 

If you want to experiment, the actual split should be trivial - you only have to replace the timeslot_handler with a module emulating its behavior (like starting/ending timeslots and passing HW IRQs to their respective handlers). There might be a couple of additional things lying around the code, but they should be easy to remove or isolate. 

The DFU-bootloader will run without the Softdevice, and will provide some reference for how you can decouple the mesh from SD.

daviddedwin, 2016-02-05:
A usable mesh DFU has been released in 0.8.0, that can be used as a basis to examine a SD less mesh.

closed 2016-02-05 by daviddedwin

olsky, 2016-06-28:
As of ver 0.8.4 I see the elegant DFU app which does not use SD (and the nice split in the makefile). Would it be now an option for the SD-less variant of the mesh? :-)