NordicPlayground/nRF51-ble-bcast-mesh #53: Turning off the mesh

matheusrfdesign, 2015-11-19:
In my project I need the mesh to be completely disabled in order to save energy. I managed decrease the current consumption down to 200uA when the mesh is inoperative by using the following function: 

	void timeslot_handler_deinit(void)
	{
		if (g_framework_initialized)
		{
			uint32_t error;
			g_is_in_callback = true;
			g_framework_initialized = false;
			error = sd_radio_session_close();
			APP_ERROR_CHECK(error);
			radio_disable();
		}
	}

However, the consumption jumps back to 500uA a few milliseconds after the mesh has been deactivated. I assume that there might be some timer event turning the radio back on. So I have been wondering if there is a better way to deactivate all the resources used by the mesh in order to improve the energy consumption.

trond-snekvik, 2015-11-19:
Hi, have you tried the [`rbc_mesh_stop()` function call](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/rbc_mesh.h#L180)? It should accomplish what you want, without having the mesh trying to bring itself back from the dead. When you order radio_session_close(), you'll get an event to the timeslot_handler  saying SESSION_CLOSED, and the mesh propagates an error to the application. When you do it without closing the current timeslot, you may experience that the mesh attempts to turn itself back on.

matheusrfdesign, 2015-11-19:
Hi, thanks for the answering! Although, I forgot to say that I am still using the older version of the mesh, so the rbc_mesh_stop() call is not available.

trond-snekvik, 2015-11-20:
Ah, I see. I would recommend copying the approach from the implementation of `rbc_mesh_stop()`. It's all contained in [this commit](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/commit/973c58e46339b3b3f6c60c1bb2ac87939af6ce43), in the [timeslot_handler.c-file](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/973c58e46339b3b3f6c60c1bb2ac87939af6ce43/nRF51/rbc_mesh/src/timeslot_handler.c). Notice how it doesn't stop the session, but just [cancels the current timeslot without ordering a new one](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/973c58e46339b3b3f6c60c1bb2ac87939af6ce43/nRF51/rbc_mesh/src/timeslot_handler.c#L237)

closed 2016-02-05 by daviddedwin