NordicPlayground/nRF51-ble-bcast-mesh #55: Cannot increase handle count and have characteristics with both and Read and Write on BLE_Gateway example.

rifatmahmud, 2015-11-20:
Has the latest change removed the previous features of adding handles and having both read and write capability on each handle? If it is changed, please, suggest a way how to get the old feature back.

trond-snekvik, 2015-11-20:
Yeah, this comes down to the new cache solution. The read operation is still available, but you'll risk getting the `NRF_ERROR_NOT_FOUND` if it's not in the cache. 

If you want to mitigate this, you have two options:

1. Increase your cache size by changing the [`RBC_MESH_DATA_CACHE_ENTRIES` #define](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/rbc_mesh.h#L67).
2. Mark some handles as "persistent", meaning that they will never be flushed from the cache. If you have assigned a few selected handles to each device, this is the best option.

If you have a cache size larger than the range of handles you're using, the behavior will be exactly as before.

rifatmahmud, 2015-11-20:
Can you please point me to the specific of this cache solution? Is this
change applicable to the whole mesh service?

rifatmahmud, 2015-11-23:
Got the explanations on the release notes. Should have looked at there first. My bad. :)

closed 2015-11-23 by rifatmahmud

trond-snekvik, 2015-11-23:
Hi, sorry about the long response time on this last issue, let us know if you have any further questions :)