NordicPlayground/nRF51-ble-bcast-mesh #56: BLE_Gateway dsconnect (timeout)

olsky, 2015-11-23:
Hi, I notice constant disconnect, when using BLE_Gateway. In my test I have an app displaying all messages received from mesh. min_inteval is set to 100 as in example. 

Could you please suggest where to hunt? :_)

thank you!
Cheers 
Oleh

trond-snekvik, 2015-11-23:
Hi, that does sound a bit odd. Are you certain that your device isn't crashing? There seems to be some instabilities in the timeslot-ordering on the latest version, make sure you haven't entered the Hardfault- or App_error-handler. If it isn't the mesh, It might be a implementation issue with the BLE-gateway's interaction with the softdevice.

A couple of questions to pinpoint this:
What's your connection interval?
Are you using the standard Gateway-mesh service, or do you relay the data in other ways?
What phone-OS are you using?

olsky, 2015-11-23:
Hi Trond,

yes it is very often in the Hardfault, with the latest version. I could not investigate it yet, but will attach and provide more details. after flashing new version it seems to reboot with the frequency of 1 to few minutes. 

Also, I have not yet applied your fix with "75ppm LFCLK", will do it as well later.

The parameters are all default as from the example (unchanged). And I use standard gateway mesh service, the data are produced but buttons. Nothing special here.

Previously, I had a timer running to test the throughput, just sending messages with the predefined frequency -- then I noticed reboots. So to isolate I removed all custom code.

The app runs on Android, connected to one node and subscribed on notify. Could you please suggest the connection range parameters  (latency, timeout e tc)? Another behaviour when connected: not all packages land on the connected node, but I think it can be explained (because of the connection the timeslot has not much room to catch all messages?).

olsky, 2015-11-23:
forgot to mention: 
- I see the disconnect (reason timeout or similar), when using MasterControl and connecting via nRF dongle.
- from app on android just disconnect or  with timeout

trond-snekvik, 2015-11-23:
We've gotten similar feedback from others on v0.7.0, and I've been looking into reasons for the crashes. I think I've found and removed them all by now, but I will do some extra system tests to increase the confidence level. I'll put it into a v0.7.1 today or first thing tomorrow (EU time), I think this should solve all the issues you're experiencing. The delta is a little too big for a hotfix, sorry for the inconvenience.

For the connection parameters, the mesh should be able to handle any BLE spec-legal timing, but I would recommend not going below 15ms on the connection interval, as the mesh goes into a sort of panic mode, choosing a different tactic for ordering timeslots, which may result in poor utilization. The connection interval also introduces a tradeoff between mesh-utilization and connection-response time, as you mention. In general, the SD seems to require ~5ms per connection event, and the mesh radio utilization-percentage can be calculated from this and the connection interval :)

olsky, 2015-11-24:
Awesome! You guys rock! :+1:

trond-snekvik, 2015-11-24:
[It's up](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/releases/tag/v0.7.1-SD8), give it a spin:)

olsky, 2015-11-24:
Sure I will! :-)

Thanks! great job :+1:

olskyy, 2015-11-26:
Great! Nodes are running without any reboot. 

Thumbs up! :+1:

closed 2016-02-05 by daviddedwin