NordicPlayground/nRF51-ble-bcast-mesh #57: Adding rbc_mesh to ble_app_gls example.

rifatmahmud, 2015-11-26:
I have copied the rbc_mesh folder on the top folder, given the reference on makefile. Then, I followed your direction in the main.c on adding the rbc_mesh.
But when I run the make command, [this](https://gist.github.com/rifatmahmud/9f948db883f808a7cb52) error message is shown.
How to resolve this issue?

trond-snekvik, 2015-11-26:
Okay, quick checklist before we proceed to make sure you didn't forget anything:

- Did you remember to add mesh_packet.c to the C_SOURCE_FILES in your makefile? 
- Is the mesh_packet.h file in one of the include paths? 
- Do you have another file called mesh_packet.h, or have you defined _MESH_PACKET_H__ somewhere else for any reason?
- Can you see the mesh_packet.o file in your compiler output directory (<example>/gcc/_build in the examples)?
- Did you get any other compiler errors before the ones in the gist?

rifatmahmud, 2015-11-27:
Thanks. Adding the mesh_packet.c in Makefile fixed the issue. But after compiling, I cannot find the mesh from my mobile's nRF Master Control Panel app.

I copied the ble_app_gls as ble_app_custom and changed the source to add a custom service of mine and removing battery and gls service, only utilizing the pairing functionality of the app.

[Here](https://gist.github.com/rifatmahmud/f8c8a4cb3a4360e1c3bd) is the main.c. Listed the rbc_mesh adding related modification below. Note that, the mesh_start is the conditional variable for starting the mesh service.

1. On [line 168](https://gist.github.com/rifatmahmud/f8c8a4cb3a4360e1c3bd#file-main-c-L168), added the rbc_mesh_event_handler function.
2. On [line 678](https://gist.github.com/rifatmahmud/f8c8a4cb3a4360e1c3bd#file-main-c-L678) added the mesh to evt_dispatch.
3. On [line 698](https://gist.github.com/rifatmahmud/f8c8a4cb3a4360e1c3bd#file-main-c-L698) added mesh_sys_evt.
4. On [line 1005](https://gist.github.com/rifatmahmud/f8c8a4cb3a4360e1c3bd#file-main-c-L1005) started rbc_mesh in main function.

[Here](https://gist.github.com/rifatmahmud/789ecf295ce28cbe6f06) is the modified nrf_adv_conn.c. Modifications are-

1. On [line 144](https://gist.github.com/rifatmahmud/789ecf295ce28cbe6f06#file-nrf_adv_conn-c-L144) commented out the ble_sd_enable

[Here](https://gist.github.com/rifatmahmud/5db7b67320fc9c375788) is the modified mesh_gatt.c. Modifications are-

1. On [line 217](https://gist.github.com/rifatmahmud/5db7b67320fc9c375788#file-mesh_gatt-c-L217)  changed the security level for MITM protection.
2. On [line 297](https://gist.github.com/rifatmahmud/5db7b67320fc9c375788#file-mesh_gatt-c-L297) changed the security level for MITM protection.

Here is the [ble_app_custom](https://drive.google.com/file/d/0B-J_8-KYz50mOFFlN19MYjFCQms/view?usp=sharing) folder with modified files. You can compile it with gcc putting it in examples/ble_peripheral folder.

trond-snekvik, 2015-11-27:
Thanks for the detailed report. I ran through your program with the JLink GDB server connected, and discovered some issues with both the mesh and the program:
1. The `err_code` at main.c:321 must be initialized, I got an app error at main.c:330, because the error_code just contained stack-garbage.
2. The timeslot SD-event handler doesn't take too kindly on irrelevant soc-events from the SD. This is a problem with the rbc_mesh, and I've created [an issue for it](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/issues/58). As a hotfix, change timeslot.c:183 from `APP_ERROR_CHECK(NRF_ERROR_INVALID_STATE)` to `break;`.
3. The `error_code` in nrf_adv_conn.c:140 has the same issue as the one in timer_init, when the call to `sd_ble_enable()` is commented out. Either comment out the error check, or initialize the `error_code` variable.

By fixing these errors, I made your code run, and it shows up in the Master control panel :)

rifatmahmud, 2015-11-28:
Thanks a lot! :+1: 

Just before closing the issue, I need to clarify one thing.

When I want to compile for PCA10031 by only changing the .ld file to [this](https://gist.github.com/rifatmahmud/4cc8746f65ab69c1269e), I get this error message while linking:

Linking target: nrf51422_xxac_s110.out
/usr/local/gcc-arm-none-eabi-4_8-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
collect2: error: ld returned 1 exit status
make: *** [nrf51422_xxac_s110] Error 1

Is it objectively not possible to run this example on PCA10031 Board(in my case it is an [RBL nRF51822](http://redbearlab.com/redbearlab-nrf51822/))?

trond-snekvik, 2015-11-28:
Yeah, looks like your config needs more than 16kB of RAM. You should be able to reduce it quite a bit by changing some configurations. The assembly-startup file from the SDK defines the size of your STACK and HEAP. If you don't use malloc, you can safely set HEAP size to 0. You might be able to reduce stack size as well, but you would have to experiment a bit. Buffer sizes in the framework are also up for reduction. In particular, the caches in version_handler.h can be reduced to 10% of their sizes.

rifatmahmud, 2015-12-01:
Sorry, had to reopen the issue.

[Here](https://gist.github.com/rifatmahmud/789ecf295ce28cbe6f06#file-nrf_adv_conn-c-L76), I have changed the gap params to have bond and mitm. Also, I had made the security level to have mitm protection [here](https://gist.github.com/rifatmahmud/5db7b67320fc9c375788#file-mesh_gatt-c-L217) and [here](https://gist.github.com/rifatmahmud/5db7b67320fc9c375788#file-mesh_gatt-c-L297). Still, when I make mesh_start = 1 in [main.c](https://gist.github.com/rifatmahmud/f8c8a4cb3a4360e1c3bd#file-main-c-L1005), the bonding operation [not showing the passkey], as it does in case I make [mesh_start = 0](https://gist.github.com/rifatmahmud/f8c8a4cb3a4360e1c3bd#file-main-c-L135).

[Here] (https://drive.google.com/file/d/0B-J_8-KYz50mcldvSUVvajVIZmM/view?usp=sharing) is the changed code.

rifatmahmud, 2015-12-01:
To see the displayed passkey, you have to open Tera Term on the COM port of nRF51 DK, with baud rate 38400, flow control none, parity none and stop bit 1.

rifatmahmud, 2015-12-07:
Hi @trond-snekvik , do you have any suggestion?

trond-snekvik, 2015-12-07:
Hi, sorry, I haven't had time to look at it yet. If I understand you correctly, I think the [write-authentication](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/src/mesh_gatt.c#L301) might be what you're looking for. You will get [a auth-event](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s110.api.v8.0.0/group___b_l_e___g_a_t_t_s___w_r_i_t_e___r_e_q___a_u_t_h___m_s_c.html) on each read or write request, which you have to authorize. It's not the one-time authorization you might have been looking for, but it should solve your problem.

closed 2016-03-03 by rifatmahmud