NordicPlayground/nRF51-ble-bcast-mesh #59: Mesh access address length and uniqueness.

rifatmahmud, 2015-12-01:
What is the max allowed length of mesh access address and how many unique possibilities of Mesh Access Address is there?

trond-snekvik, 2015-12-01:
Hi, the Mesh access address is bound to 4 bytes. The access address is defined in the BLE specification, and allows us to filter packets in hardware. The BLE spec defines access address 0x8E89BED6 for regular advertisement packets, but we've left it to the application to define their own, as it might be beneficial to separate all mesh-traffic from all advertisements in your environment. If I remember correctly, all example projects use the spec-defined access address above, so that the packets are visisble on packet sniffers and standard BLE devices. 0x8E89BED6 is defined as `RBC_MESH_ACCESS_ADDRESS_BLE_ADV` in rbc_mesh.h. 

While the specifications have some rather cumbersome rules to the access address (things like number of successive bits being the same, and number of bits difference from the standard access address), we're not enforcing these rules in the framework, and you may use any of the 2^32 different access addresses.

trond-snekvik, 2015-12-01:
It is worth mentioning that the access-address rules are there for a reason though; the access address acts as a "sync" for the radio hardware, making it easier to lock onto an incoming message and separate it from general noise. Having a poor access address (like 0x00000000 or some other lazy number) will lead to more corrupted packets and lower throughput, as you're making the radio hardware's life harder. 

If you're interested, the BLE specification's rules are defined in Core spec 4.2 Vol. 6 Part B Chapter 2.1.2 and are as follows:
> The initiator shall ensure that the Access Address meets the following requirements:
> • It shall have no more than six consecutive zeros or ones.
> • It shall not be the advertising channel packets’ Access Address.
> • It shall not be a sequence that differs from the advertising channel packets’ Access Address by only one bit.
> • It shall not have all four octets equal.
> • It shall have no more than 24 transitions.
> • It shall have a minimum of two transitions in the most significant six bits.

closed 2016-03-03 by rifatmahmud