NordicPlayground/nRF51-ble-bcast-mesh #60: Another instance - feasible?

olsky, 2015-12-03:
Hi, in some earlier versions I do recall there was a possibility to start more instances, is it still the case?
Was this tried/tested? 
I thought to start two independent instances on different channels.

Can you please share some infos?

trond-snekvik, 2015-12-03:
Could you elaborte a bit? At which level of instancing?

olsky, 2015-12-03:
sure, wanted to try 3 independed mesh instances on different radio channels (with own name-value map and own tickle instance).

trond-snekvik, 2016-01-04:
Hi, sorry for the delay.
I'm afraid you can't start multiple concurrent instances of the mesh on a single chip. I think what you're referring to was meant to be multiple network-instances in a single location, but I don't think the documentation was clear enough. This is of course still possible by setting different channels/access addresses on different devices, but a single device will be restricted to a single configuration at a time in the base mesh-implementation. 

If you're still interested in doing multiple instances, there's a way to do this with access addresses in the hardware, but it will require quite extensive additions throughout the stack. I can walk through them for you if you're up for it, but I doubt it will be part of the main version of the mesh-framework, sorry.

olsky, 2016-02-18:
Thanks Trond!
Yes, I'm interested! :) sounds really cool...

trond-snekvik, 2016-03-01:
To do this, we need to change the radio configuration, and expose this change further up the stack.

*First, multiple access addresses:*

The nRF51 radio has a limited number of access addresses it can match packets to. These addresses are set in the `NRF_RADIO->PREFIXn` and `NRF_RADIO->BASEn` registers. In our stack, these are configured in [`radio_init()`](../blob/sdk-8-support/nRF51/rbc_mesh/src/radio_control.c#L165):

```
    /* Configure Access Address  */
    NRF_RADIO->PREFIX0	    = ((access_address >> 24) & 0x000000FF);
    NRF_RADIO->BASE0 		= ((access_address << 8) & 0xFFFFFF00);
    NRF_RADIO->TXADDRESS    = 0x00;			    // Use logical address 0 (prefix0 + base0) = 0x8E89BED6 when transmitting
    NRF_RADIO->RXADDRESSES  = 0x01;				// Enable reception on logical address 0 (PREFIX0 + BASE0)
```

As you can see, we define PREFIX and BASE 0. You can only have two different base addresses, and 8 different prefixes, but you can't mix and match as you want: Table 99 in the [nRF51 Reference manual v3.0 (pdf!)](http://www.nordicsemi.com/eng/nordic/download_resource/20337/12/39837117) defines 8 possible combinations. These indexes defined for each combination is to be used in the `NRF_RADIO->TXADDRESS` and `NRF_RADIO->RXADDRESSES` registers. The `TXADDRESS` is an index (matching the table), the `RXADDRESSES` is a bitfield, where you can mark which access addresses you want to receive on. After [receiving a packet](../blob/sdk-8-support/nRF51/rbc_mesh/src/radio_control.c#L282), you can get the index of the address-combination you received on in the `NRF_RADIO->RXMATCH` register.

*Multiple channels:*

Channels are a bit easier to deal with than access addresses. Setting the hardware-channel only requires that you set `NRF_RADIO->FREQUENCY` to the frequency of the BLE channel you want, and setting `NRF_RADIO->DATAWHITEIV` to the channel index you want. Luckily, this mapping is already done in [`radio_channel_set(uint8_t ch)`](../blob/sdk-8-support/nRF51/rbc_mesh/src/radio_control.c#L118), so you only need to store the BLE channel, and set it correctly in `setup_event(radio_event_t* p_evt)`.

*Passing the parameters:*

Configuring a second access address or channel should be as easy as duplicating the access address or channel parameter all the way from the API (or hardcode it, if you won't bother:) ). Binding the access address or channel to the packets you're sending requires more work. I don't know your use-case, but you could either split your handle range into multiple configurations by some global rule, or find a smart way to specify (and store!) them throughout the stack. Sending on the correct `TXADDRESS` and channel should be solvable by adding extra parameters to the [`radio_event_t` struct](../blob/sdk-8-support/nRF51/rbc_mesh/include/radio_control.h#L52).

olsky, 2016-03-01:
OMG! :-) you simply rock, Trond!

That's what I was thinking of, but I was not sure if it was feasible (espacially with logical addresses -- I might have overlooked some docs, I will check the RefManual again, just in case -- could you please point to some other nRF docs?)

Thank you alot!

PS: now it gets tricky :-) I simply need some time... that's all :)))

trond-snekvik, 2016-03-03:
The three docs that are always open on my computer are 

- [nRF51 Reference manual](http://www.nordicsemi.com/eng/nordic/download_resource/20337/12/85564726): Register overview, HW functionality, module overview

- [Softdevice Specification S110](http://www.nordicsemi.com/eng/nordic/download_resource/20338/15/41758462): Softdevice S110 resource requirements, timing, API details.

- [nRF51822 Product specification](http://www.nordicsemi.com/eng/nordic/download_resource/20339/13/89629302): Power consumption, HW module instances, HW timing.

All info relevant for radio config is in the RM (Section 17) :)

olsky, 2016-03-03:
Trond: thank you! 

I must admit ref-man and prod-spec are my pref read. I must force myself into S110 spec reading...