NordicPlayground/nRF51-ble-bcast-mesh #67: Restarting the mesh with a new mesh access address inside a program.

rifatmahmud, 2016-01-29:
Does calling rbc_mesh_init() with new init_param(i.e. mesh access address) 2nd time restart the mesh with new access adress?

If not, is there a way to do it?

trond-snekvik, 2016-03-01:
Hi, sorry for the delay. You can't call rbc_mesh_init() again, but you can change the parameter in transport control's function `tc_radio_params_set(uint32_t access_address, uint8_t channel)`. This will trigger a reset of the radio, and will cause your device to change access address within a couple of milliseconds.

rifatmahmud, 2016-03-03:
Can I call it from say, main.c(by including transport_control.h) without any problem?

trond-snekvik, 2016-03-03:
That should be fine, but you'll have to find a way to synchronize it across the mesh if you want to have all devices switching. They won't be able to relay messages on the old access address once you switch (unless you switch back of course)