NordicPlayground/nRF51-ble-bcast-mesh #69: Mesh using NRF51

junaediadi, 2016-02-17:
Dear All,
Hello, I will develop bluetooth Mesh and found that in this project they used NRF51.
Could you help me,what is the limitation of this project,and do we need smartphone like in CSR as a central device?

Thanks,

daviddedwin, 2016-02-17:
The Smartphone can participate in the Mesh using GATT (Generic Attribute Profile) but is not a required part of the Mesh.

junaediadi, 2016-02-17:
Could you help me to mention,what equipments/devicesthat I need to build this system beside the NRF51 DK?

daviddedwin, 2016-02-17:
The nRF51-DK or nRF51-Dongle is sufficient to build and use the Mesh, you need to have 2 or more of them.

junaediadi, 2016-02-17:
Dear David,

Thanks for your explanation. SO nRF51DK act also as a central device to send command to other device? is it right?

daviddedwin, 2016-02-17:
The nRF51 DK can act as a peripheral, central and as a mesh node. This article can help (it is a but old) https://devzone.nordicsemi.com/blogs/672/accessing-the-ble-mesh-via-the-spark-cloud/ , but can act as a useful starting point.

junaediadi, 2016-02-17:
Dear David,
Thanks for your help.I will start to learn from that starting point..

Br,
Junaedi Adi P

closed 2016-02-17 by junaediadi