NordicPlayground/nRF51-ble-bcast-mesh #73: DFU update with different Firmware/Hardware

olsky, 2016-03-09:
Hi,

I was wondering if the current DFU will support the following scenario: suppose there are 2 variants of hardware, so these require separate firmware.

Lets start the update of FW-1, will the devices HW-2 relay packets intended for the HW-2? And then other way around?

So that in two updates one can update the whole network?

Thanks!

trond-snekvik, 2016-03-10:
Hi Olsky, great question!

Yes, this is possible :) When in bootloader, all devices will attempt to participate as relays for transfers they can't be receivers of. The framework doesn't yet support relay-functionality while in application, but we hope to add this in the future, along with DFU-source functionality. The current functionality of the lower layers of the framework isn't able to support this as of today, as it is very locked into serving the specific transmit patterns of the upper layers.

olsky, 2016-03-10:
Hi Trond, thanks for your reply.

It was the natural question as I have now another lot of custom boards with 32K (the previous was still 16K), so I have a separate hex file for them, as these have different MCUs and different hw-configuration.

So I need to put all devices into DFU mode, and then at the end of the update process recover all of them into app mode?

With the latest release all works very stable and no issues at all :)

trond-snekvik, 2016-03-10:
Yes :)

closed 2016-04-06 by trond-snekvik