NordicPlayground/nRF51-ble-bcast-mesh #76: Scaling_example gcc build missing rand.c, bootloader

trjw, 2016-03-15:
Using the latest sdk-8-support branch, doing a gcc build of Scaling_example fails due to some (apparently) missing fragments from gcc/makefile:

I'm very early in the BLE/mesh learning curve so this is just copy-pasta from the BLE_Gateway example, however it does at least allow the build to complete.

    diff --git a/nRF51/examples/Scaling_example/gcc/makefile b/nRF51/examples/Scaling_example/gcc/makefile
    index 076fbde..3c4b974 100644
    --- a/nRF51/examples/Scaling_example/gcc/makefile
    +++ b/nRF51/examples/Scaling_example/gcc/makefile
    @@ -89,6 +89,11 @@ C_SOURCE_FILES += ../../../rbc_mesh/src/fifo.c
     C_SOURCE_FILES += ../../../rbc_mesh/src/event_handler.c
     C_SOURCE_FILES += ../../../rbc_mesh/src/version_handler.c
     C_SOURCE_FILES += ../../../rbc_mesh/src/mesh_packet.c
    +C_SOURCE_FILES += ../../../rbc_mesh/src/bootloader_app.c
    +C_SOURCE_FILES += ../../../rbc_mesh/src/bootloader_info_app.c
    +C_SOURCE_FILES += ../../../rbc_mesh/src/bootloader_util.c
    +C_SOURCE_FILES += ../../../rbc_mesh/src/rand.c
    +

John

trond-snekvik, 2016-03-15:
Hi, sorry about this one. It slipped through the cracks in the 0.8 release. Thanks for reporting, I'll try to push the change into the next release.

trond-snekvik, 2016-03-15:
Your fixes are valid, and should not cause any other issues (apart from your #77, but that's somewhat unrelated)

trond-snekvik, 2016-06-30:
The Scaling example has been deprecated in 0.8.4.

closed 2016-06-30 by trond-snekvik