NordicPlayground/nRF51-ble-bcast-mesh #77: sdk-8-support ping_pong example fails to build

trjw, 2016-03-15:
Hi,

The ping_pong example (gcc build) in latest sdk-8-support branch fails to build:

```
Linking target: ping_pong.elf
_build/trickle.o: In function `refresh_t':
/home/jwilliams/Documents/nordic/sdkv8/examples/nRF51-ble-bcast-mesh/nRF51/examples/ping_pong/gcc/../../../rbc_mesh/src/trickle.c:81: undefined reference to `rand_prng_get'
_build/trickle.o: In function `trickle_setup':
/home/jwilliams/Documents/nordic/sdkv8/examples/nRF51-ble-bcast-mesh/nRF51/examples/ping_pong/gcc/../../../rbc_mesh/src/trickle.c:113: undefined reference to `rand_prng_seed'
_build/transport_control.o: In function `mesh_framework_packet_handle':
/home/jwilliams/Documents/nordic/sdkv8/examples/nRF51-ble-bcast-mesh/nRF51/examples/ping_pong/gcc/../../../rbc_mesh/src/transport_control.c:281: undefined reference to `bootloader_start'
/home/jwilliams/Documents/nordic/sdkv8/examples/nRF51-ble-bcast-mesh/nRF51/examples/ping_pong/gcc/../../../rbc_mesh/src/transport_control.c:288: undefined reference to `bootloader_start'
/home/jwilliams/Documents/nordic/sdkv8/examples/nRF51-ble-bcast-mesh/nRF51/examples/ping_pong/gcc/../../../rbc_mesh/src/transport_control.c:302: undefined reference to `bootloader_start'
/home/jwilliams/Documents/nordic/sdkv8/examples/nRF51-ble-bcast-mesh/nRF51/examples/ping_pong/gcc/../../../rbc_mesh/src/transport_control.c:310: undefined reference to `bootloader_start'
_build/transport_control.o: In function `tc_init':
/home/jwilliams/Documents/nordic/sdkv8/examples/nRF51-ble-bcast-mesh/nRF51/examples/ping_pong/gcc/../../../rbc_mesh/src/transport_control.c:328: undefined reference to `bootloader_info_init'
/home/jwilliams/Documents/nordic/sdkv8/examples/nRF51-ble-bcast-mesh/nRF51/examples/ping_pong/gcc/../../../rbc_mesh/src/transport_control.c:329: undefined reference to `bootloader_info_entry_get'
collect2: error: ld returned 1 exit status
make: *** [all] Error 1

```

I added the rand.c and various bootloader objects to the makefile similar to my issue report for Scaling_example #76, however it fails now in a slightly different way:

```
Compiling file: ../../../rbc_mesh/src/bootloader_util.c
../../../rbc_mesh/src/bootloader_util.c: In function 'bootloader_util_reset':
../../../rbc_mesh/src/bootloader_util.c:124:1: error: r7 cannot be used in asm here
 }

```

I haven't dug through the uView build files to see what I've missed in this bit of copy-pasta.

John

trond-snekvik, 2016-03-15:
This is caused by gcc using r7 as frame pointer if you don't run with optimization. Does adding `-fomit-frame-pointer` to your compiler options solve this?

trjw, 2016-03-17:
Yep that fixes the build thanks.  Haven't tried running it yet.

```
diff --git a/nRF51/examples/ping_pong/gcc/makefile b/nRF51/examples/ping_pong/gcc/makefile
index 9a9638c..d030679 100755
--- a/nRF51/examples/ping_pong/gcc/makefile
+++ b/nRF51/examples/ping_pong/gcc/makefile
@@ -90,6 +90,11 @@ C_SOURCE_FILES += ../../../rbc_mesh/src/fifo.c
 C_SOURCE_FILES += ../../../rbc_mesh/src/event_handler.c
 C_SOURCE_FILES += ../../../rbc_mesh/src/version_handler.c
 C_SOURCE_FILES += ../../../rbc_mesh/src/mesh_packet.c
+C_SOURCE_FILES += ../../../rbc_mesh/src/bootloader_app.c
+C_SOURCE_FILES += ../../../rbc_mesh/src/bootloader_info_app.c
+C_SOURCE_FILES += ../../../rbc_mesh/src/bootloader_util.c
+C_SOURCE_FILES += ../../../rbc_mesh/src/rand.c
+
 ASM_SOURCE_FILES += gcc_startup_nrf51.s # using local startup file to cater customized heap size
 
 C_SOURCE_FILES += $(COMPONENTS)/ble/common/ble_advdata.c
@@ -158,6 +163,7 @@ CFLAGS += -mfloat-abi=soft
 CFLAGS += -ffunction-sections 
 CFLAGS += -fdata-sections -fno-strict-aliasing
 CFLAGS += -fno-builtin
+CFLAGS += -fomit-frame-pointer
 
 # Linker flags
 LDFLAGS += -Xlinker -Map=$(LISTING_DIRECTORY)/$(OUTPUT_NAME).map
```

trond-snekvik, 2016-12-01:
Closed due to inactivity. Reopen if needed.

closed 2016-12-01 by trond-snekvik