NordicPlayground/nRF51-ble-bcast-mesh #80: Strange code in transport_control.c

victorpasse, 2016-03-24:
int16_t delta = vh_get_version_delta(p_mesh_adv_data->handle, p_mesh_adv_data->version);

This makes no sense, p_mesh_adv_data->handle has nothing to do with the old version.
Is this a mistake or am I confused?

victorpasse, 2016-03-24:
I would suggest editing vh_get_version_delta such that it takes an handle and a new version as there are some static functions in version handler C file that extracts version from handle that can still be static.

trond-snekvik, 2016-03-29:
This is a mistake, you're correct. I think the cleanest fix is to add a return-by-pointer argument to the `vh_rx_register(p_mesh_adv_data, timestamp)` call right below it, as it calculates the delta anyway.

closed 2016-06-27 by trond-snekvik