NordicPlayground/nRF51-ble-bcast-mesh #82: Inconsistent ble_adv_addr in rbc_mesh_event_t

olsky, 2016-03-29:
I see some inconsistency in 

rbc_mesh_event_handler(rbc_mesh_event_t* evt)

first 3 bytes of the address are ok, but not the last three. 
Test scenario:

- 2 nodes
- on startup I dump the address using: sd_ble_gap_address_get
- in rbc_mesh_event_handler, dump ble_adv_addr 
- the address is not matching

Am I missing something? Thank you.

trond-snekvik, 2016-03-29:
https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/src/transport_control.c#L112 ? :)

olsky, 2016-03-29:
yes, I've seen it, and have no clue why it can be different :) 
decided to check if its only at my side...

trond-snekvik, 2016-03-29:
Essentially, the line `memcpy(&evt->ble_adv_addr, p_addr, sizeof(p_mesh_adv_data));` copies only four bytes of the address structure, since `sizeof(<a pointer>)` is 4 on a 32-bit system. 

The `ble_gap_addr_t`-structure we're copying is defined in the SD-header ble_gap.h, Line 394:

```
typedef struct
{
  uint8_t addr_type;                    /**< See @ref BLE_GAP_ADDR_TYPES. */
  uint8_t addr[BLE_GAP_ADDR_LEN];       /**< 48-bit address, LSB format. */
} ble_gap_addr_t;
```
and uses the first byte as addr type. Therefore, only 3 bytes of the addr-array are copied. Change the line to `memcpy(&evt->ble_adv_addr, p_addr, sizeof(ble_gap_addr_t));`, and you should see some improvements :)

olsky, 2016-03-29:
:-) thanks, have had a coredump somehow :-D

closed 2016-03-29 by olsky