NordicPlayground/nRF51-ble-bcast-mesh #83: More repeats than anticipated in mesh

victorpasse, 2016-03-30:
Hi, I am doing testing of the mesh in a 16 unit setup where all units can see all units.
I have sent data on the mesh on several indices and then stop sending data.
I would expect that all indices would be repeated atleast K=3 times and atmost about (Number of units)+K = 19 times after no more updates are sent. I get the following results several minutes after not sending data:
`Pkg- I: 0x41, Ver: 0x234, Delta: 0, Raw Len: 14;
Pkg- I: 0xAD, Ver: 0x1F7, Delta: 0, Raw Len: 14;
Pkg- I: 0xB9, Ver: 0x221, Delta: 0, Raw Len: 14;
Pkg- I: 0xB9, Ver: 0x221, Delta: 0, Raw Len: 14;
Pkg- I: 0x44, Ver: 0x1A7, Delta: 0, Raw Len: 14;
Pkg- I: 0xAD, Ver: 0x1F7, Delta: 0, Raw Len: 14;
Pkg- I: 0xA4, Ver: 0x21D, Delta: 0, Raw Len: 14;
Pkg- I: 0x41, Ver: 0x234, Delta: 0, Raw Len: 14;
Pkg- I: 0x41, Ver: 0x234, Delta: 0, Raw Len: 14;
Pkg- I: 0x25, Ver: 0x26, Delta: 0, Raw Len: 14;
Pkg- I: 0x25, Ver: 0x26, Delta: 0, Raw Len: 14;
Pkg- I: 0x25, Ver: 0x26, Delta: 0, Raw Len: 14;
Pkg- I: 0xAC, Ver: 0x1FA, Delta: 0, Raw Len: 14;
Pkg- I: 0xAC, Ver: 0x1FA, Delta: 0, Raw Len: 14;
Pkg- I: 0x16, Ver: 0x2DA2, Delta: 0, Raw Len: 14;
Pkg- I: 0x16, Ver: 0x2DA2, Delta: 0, Raw Len: 14;
Pkg- I: 0x41, Ver: 0x234, Delta: 0, Raw Len: 14;
Pkg- I: 0x41, Ver: 0x234, Delta: 0, Raw Len: 14;
Pkg- I: 0xA5, Ver: 0x21E, Delta: 0, Raw Len: 14;
Pkg- I: 0xA5, Ver: 0x21E, Delta: 0, Raw Len: 14;
Pkg- I: 0x4D, Ver: 0x5034, Delta: 0, Raw Len: 14;
Pkg- I: 0x4D, Ver: 0x5034, Delta: 0, Raw Len: 14;
Pkg- I: 0xAA, Ver: 0xAD5D, Delta: 0, Raw Len: 14;
Pkg- I: 0x4, Ver: 0x9035, Delta: 0, Raw Len: 14;
Pkg- I: 0xAA, Ver: 0xAD5D, Delta: 0, Raw Len: 14;
Pkg- I: 0xAA, Ver: 0xAD5D, Delta: 0, Raw Len: 14;
Pkg- I: 0xBE, Ver: 0x43, Delta: 0, Raw Len: 14;
Pkg- I: 0x4B, Ver: 0x7E09, Delta: 0, Raw Len: 14;
Pkg- I: 0x53, Ver: 0xE73A, Delta: 0, Raw Len: 14;
Pkg- I: 0xB6, Ver: 0xD4DC, Delta: 0, Raw Len: 14;
Pkg- I: 0xB6, Ver: 0xD4DC, Delta: 0, Raw Len: 14;
Pkg- I: 0xB6, Ver: 0xD4DC, Delta: 0, Raw Len: 14;
Pkg- I: 0xB6, Ver: 0xD4DC, Delta: 0, Raw Len: 14;
Pkg- I: 0xAA, Ver: 0xAD5D, Delta: 0, Raw Len: 14;
Pkg- I: 0x4B, Ver: 0x7E09, Delta: 0, Raw Len: 14;
Pkg- I: 0x3F, Ver: 0xCAD6, Delta: 0, Raw Len: 14;
Pkg- I: 0x3F, Ver: 0xCAD6, Delta: 0, Raw Len: 14;
Pkg- I: 0x4, Ver: 0x9035, Delta: 0, Raw Len: 14;
Pkg- I: 0x40, Ver: 0x66D7, Delta: 0, Raw Len: 14;
Pkg- I: 0x40, Ver: 0x66D7, Delta: 0, Raw Len: 14;
Pkg- I: 0x53, Ver: 0xE73A, Delta: 0, Raw Len: 14;
Pkg- I: 0x53, Ver: 0xE73A, Delta: 0, Raw Len: 14;
Pkg- I: 0x4B, Ver: 0x7E09, Delta: 0, Raw Len: 14;
Pkg- I: 0x3F, Ver: 0xCAD6, Delta: 0, Raw Len: 14;
Pkg- I: 0x3F, Ver: 0xCAD6, Delta: 0, Raw Len: 14;
Pkg- I: 0x4B, Ver: 0x7E09, Delta: 0, Raw Len: 14;
Pkg- I: 0x4B, Ver: 0x7E09, Delta: 0, Raw Len: 14;
Pkg- I: 0x47, Ver: 0xB6DE, Delta: 0, Raw Len: 14;
Pkg- I: 0x47, Ver: 0xB6DE, Delta: 0, Raw Len: 14;
Pkg- I: 0x50, Ver: 0xA374, Delta: 0, Raw Len: 14;
Pkg- I: 0x50, Ver: 0xA374, Delta: 0, Raw Len: 14;
Pkg- I: 0x47, Ver: 0xB6DE, Delta: 0, Raw Len: 14;
Pkg- I: 0x47, Ver: 0xB6DE, Delta: 0, Raw Len: 14;
Pkg- I: 0x4D, Ver: 0x5034, Delta: 0, Raw Len: 14;
Pkg- I: 0x18, Ver: 0x6605, Delta: 0, Raw Len: 14;
Pkg- I: 0x4C, Ver: 0x26F, Delta: 0, Raw Len: 14;
Pkg- I: 0xAD, Ver: 0x1F7, Delta: 0, Raw Len: 14;
Pkg- I: 0x2F, Ver: 0xAE39, Delta: 0, Raw Len: 14;
Pkg- I: 0x20, Ver: 0xF74F, Delta: 0, Raw Len: 14;
Pkg- I: 0x43, Ver: 0xAF03, Delta: 0, Raw Len: 14;
Pkg- I: 0x50, Ver: 0xA374, Delta: 0, Raw Len: 14;
Pkg- I: 0x18, Ver: 0x6605, Delta: 0, Raw Len: 14;
Pkg- I: 0xAA, Ver: 0xAD5D, Delta: 0, Raw Len: 14;
Pkg- I: 0x2F, Ver: 0xAE39, Delta: 0, Raw Len: 14;
Pkg- I: 0x4, Ver: 0x9035, Delta: 0, Raw Len: 14;
Pkg- I: 0x25, Ver: 0x26, Delta: 0, Raw Len: 14;
Pkg- I: 0x4C, Ver: 0x26F, Delta: 0, Raw Len: 14;
[... 422 lines cut ...]
Pkg- I: 0x2B, Ver: 0x82AE, Delta: 0, Raw Len: 16;
Pkg- I: 0xB8, Ver: 0x21E, Delta: 0, Raw Len: 14;
Pkg- I: 0xB8, Ver: 0x21E, Delta: 0, Raw Len: 14;
Pkg- I: 0xB9, Ver: 0x221, Delta: 0, Raw Len: 14;
Pkg- I: 0xB9, Ver: 0x221, Delta: 0, Raw Len: 14;
Pkg- I: 0xAD, Ver: 0x1F7, Delta: 0, Raw Len: 14;
Pkg- I: 0xAD, Ver: 0x1F7, Delta: 0, Raw Len: 14;
Pkg- I: 0x50, Ver: 0xA374, Delta: 0, Raw Len: 14;
Pkg- I: 0x4, Ver: 0x9035, Delta: 0, Raw Len: 14;
Pkg- I: 0xBE, Ver: 0x43, Delta: 0, Raw Len: 14;
Pkg- I: 0x50, Ver: 0xA374, Delta: 0, Raw Len: 14;
Pkg- I: 0x25, Ver: 0x26, Delta: 0, Raw Len: 14;
Pkg- I: 0x25, Ver: 0x26, Delta: 0, Raw Len: 14;
Pkg- I: 0x11, Ver: 0x236, Delta: 0, Raw Len: 14;
Pkg- I: 0xAA, Ver: 0xAD5D, Delta: 0, Raw Len: 14;
Pkg- I: 0xAD, Ver: 0x1F7, Delta: 0, Raw Len: 14;
Pkg- I: 0x2B, Ver: 0x82AE, Delta: 0, Raw Len: 16;
Pkg- I: 0x2B, Ver: 0x82AE, Delta: 0, Raw Len: 16;
Pkg- I: 0x4F, Ver: 0xCF77, Delta: 0, Raw Len: 14;
Pkg- I: 0x4F, Ver: 0xCF77, Delta: 0, Raw Len: 14;
Pkg- I: 0x4F, Ver: 0xCF77, Delta: 0, Raw Len: 14;
Pkg- I: 0x11, Ver: 0x236, Delta: 0, Raw Len: 14;
Pkg- I: 0x11, Ver: 0x236, Delta: 0, Raw Len: 14;`
So the nodes never stop repeating the data.
I assume that C somehow gets cleared for the trickle instances. What could the problem be?

victorpasse, 2016-03-31:
This results in g_packet_refs being filled and I assume that the mesh will crash when all m_data_cache are filled. We need to fix this bug.

trond-snekvik, 2016-04-05:
Hi, sorry about the delay. This is actually by design. The handles are considered "states" in the system, and the goal is always to synchronize all states in all devices, including newcomers. To allow newcomers to align with the latest set of states, the framework keeps retransmitting the latest version of each handle with exponentially increasing intervals (up until the interval hits the upper limit of 2048 * min_interval, where they keep running on that interval forever). If a new device powers up, it will be able to read out the version and data of the handles it's interested in from these retransmissions. You can stop the retransmission of each handle individually, by receiving the TX-events, and calling `rbc_mesh_value_disable(handle)` after you think the version has been transmitted long enough. 

Under the same state-paradigm, an update to a handle will replace the previous version of the data on that handle, so there is no built-in guaranteed minimum number of retransmits, and there is no way to transmit two different handle-versions concurrently.

Please let me know if you have further questions on this topic, or if I misunderstood your question.

I realize that the state-system might be a bit different from the requirements in your case. We have been experimenting with changing from the state system - which really is "data-based", in the same manner as GATT is - to a "message based" system, where the handle is just a source address, the version is just a sequence number, and every version is treated individually, with no replacement strategy or infinite repeat. However, this would be a fundamental change to one of the core mechanics of the mesh, and we're not sure how we want to handle it.

victorpasse, 2016-04-06:
What happens if I have X slots in the data chache and send data for X handles, then all data slots are used forever.
If I then send one more handle no node will be able to accommodate the new handle in the data chache and the new data will be dropped?
Is this correct? If this is correct and by design I must say that the design has flaws. 

Also I can not use TX events as the reference counter will never reach zero as per issue #74 
But I get your point and will implement that solution if my above assumption is true.

trond-snekvik, 2016-04-06:
When you receive handle number x+1, the least recently updated handle in the cache will be replaced, to make room for the newest. So you may receive on 65k handles if you want, the version handler will just replace whichever handle is updated least recently.

I have a fix for #74, (and a lot of other issues) coming up, but I'm still working out a couple of new issues I've introduced in the process. I'll keep you updated.

victorpasse, 2016-04-06:
Ok, thank you, this sounds great!

closed 2016-12-01 by trond-snekvik