NordicPlayground/nRF51-ble-bcast-mesh #84: Mesh on several access addresses

mykhaylokohut, 2016-04-06:
Hi,

In our project we are trying to use several access addresses at same time. As far as we know, nRF51 allows listening up to 8 access addresses.

Our setup consist of one master node and several slave nodes.

We met some strange behavior when tried to make the master node to listen to two logical addresses. The algorithm we use:

1. The master node starts with two logical addresses configured and listens to both of them. The master selects logical address 1 as TX address.
2. The slave node sends connection request on access address that corresponds to masters logical address 0. The master switches its TX address to logical address 0 to reply to slave node with access address that corresponds to master's logical address 1. After TX_COMPLETE event master switches back to logical address 1.
3. The slave node switches to that received address and connects to master on master's logical address 1.

We have a setup with 1 master configured on two addresses, e.g. 0x916763AD and 0xE550C481 but we also tried some others. In nRF51 registers terms: base0 = 0x6763AD, base1 = 0x50C481, prefix0 = 0x91, prefix1 = 0xE5.

Next, we power on 15 slave nodes and they connect to master as described above with no issues. Then we power on 2 more slaves nodes and the master does not get the connection requests from this new nodes, however this new nodes get each others connection requests. In some cases this new nodes do connect to the master, but after about 30 attempts.

On the other hand, if we power on all 17 slave nodes at once, all 17 slave nodes connect OK.

Do you have any thoughts about this? 

Thanks,
Mykhaylo

trond-snekvik, 2016-04-07:
Hi Mykhaylo,

There is a certain delay from the actual radio switch from tx to rx until the TX_COMPLETE event is propagated to the application. When you get the TX_COMPLETE, the radio has already started receive mode (or potentially another TX), and it will be too late to change logical address. I don't know what your version of the radio code looks like, but switching between tx-addresses on a per-packet level could be a bit tricky. Since there's a delay from the `rbc_mesh_value_set()` call to the actual triggering of the tx-event, you have to store your preferred tx-address for that particular packet. I would suggest storing it in one of the _rfu-fields in the packet header, perhaps, then read that field when setting up the packet for transmit in radio_control.c's `setup_event()`.

Since both transmits and the tx-complete events are asynchronous, you need to control this behavior directly in the radio code itself.

trond-snekvik, 2016-04-07:
BTW, There's been a similar discussion in Issue #60, maybe you're able to get some additional perspective from those posts.

mykhaylokohut, 2016-04-07:
Hi Trond,

Thanks for fast response - we'll try and write you back.

Mykhaylo