NordicPlayground/nRF51-ble-bcast-mesh #85: Is the advertisement address in the broadcasted packet in mesh exactly the (static) MAC address of the device where the value first originated?

rifatmahmud, 2016-04-07:
The doc in the master branch states:

>  In addition, the Advertisement Address (GAP address) field provided after the packet header, does not necessarily contain the Advertisement Address of the node broadcasting the message, but rather the address of the mesh node at which the indicated version of the value-handle pair first appeared. 

The statement is not present in the sdk-8-support branch, which I am using.

I need to get the unique static MAC address of the device where the broadcasted value originated. Is it enclosed the mesh packet, as stated in the master branch doc?

trond-snekvik, 2016-04-07:
This was changed in [version 0.6.8](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/releases/tag/v0.6.8-SD8) to improve neighbor discovery, and to stay spec-compliant. You can change it back by removing the call to `mesh_packet_set_local_addr(p_packet)` on [line 294 of mesh_packet.c](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/src/mesh_packet.c#L294). Note that this will break any rssi-data from mesh packets, as the devices will start lying about their adv-address.

rifatmahmud, 2016-04-08:
> Note that this will break any rssi-data from mesh packets, as the devices will start lying about their adv-address.

That essentially means, when a node is rebroadcasting, am mobile device/another BLE acting as GAP central will not correctly recognize it, since it's advertising packet will contain address of another device.
Is this the implication of  removing the call to mesh_packet_set_local_addr(p_packet)?

trond-snekvik, 2016-04-08:
More or less.

The connectable advertisements created by the Softdevice should remain the same. Only packets with mesh payload will have the wrong advertisement address, so if you're using the advertisement addresses to get a list of your neighbor nodes (for instance by using [the transport-peek function](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/src/transport_control.c#L423)), you'll have some trouble. The mobile device will have similar issues, with the list of nearby devices being whatever mesh device keeps feeding packets to the network, but if you set up the softdevice to be connectable (like in the BLE_Gateway example), you'll see the correct address for those advertisements

rifatmahmud, 2016-04-08:
Thanks for the explanation.
My device is connectable. So, I guess, it'll be correctly recognized by scanner/central, while at the same time its mesh packets can contain the address of the source node.

closed 2016-04-08 by rifatmahmud