NordicPlayground/nRF51-ble-bcast-mesh #86: How to change the broadcast format?

JacksonLv, 2016-04-12:
Hi there ,I got a IoT board has WiFi and BLE function .Now I want to make this board as the one of the  member of the nRF BLE mesh but I can't see the rebroadcast data between the nRF51 with the capture device.I heard that I need to change xx.h in somewhere but not sure which it is.Anybody and help and it will be grateful !!:)

trond-snekvik, 2016-04-12:
Hi, when calling rbc_mesh_init(), you can give it an access address and a channel through the struct. To be able to see the packets with a regular sniffer/scanner, you have to use the standard BLE access address, and one of the channels 37, 38 or 39. The BLE access address is [defined in rbc_mesh.h as `RBC_MESH_ACCESS_ADDRESS_BLE_ADV`](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/rbc_mesh.h#L39). The packets are pushed as non-connectable advertisements, so depending on your capture device, you might have trouble detecting the packets. Keep me updated :)

JacksonLv, 2016-04-12:
thanks a lot for your warm helping .I will test it later .I have another question is where can I change the broadcast package size?As I use LightBlue to connect to one of the node of the mesh and send "00000001FFFF00" ,I saw other node just received and notify "00000001FF" ,2 byte missed. If it is because of the broadcast package size,how can I change it?

trond-snekvik, 2016-04-12:
You might want to take a look at [the docs for the GATT service](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/docs/how_it_works.adoc#gatt-service). Looking at Table 2, the characteristic takes a specific opcode-based format. Setting the value of a handle has the following format:
`OPCODE[1] | HANDLE[2] | LEN[1] | DATA[LEN]`. The opcode should be 0x00, the handle is a 16bit little endian value, the length is the length of the data-value. 

Applied to your message, you're sending 
Opcode | Handle | length | data
---------------------------------------
0x00 | 0x0000 | 0x01 | 0xFFFF00

Since your length is 0x01, your data is cut short. Try sending "00000003FFFF00", and you should see your last two bytes going through as well.

JacksonLv, 2016-04-13:
Thanks!! You solve my problem perfectly!!

closed 2016-04-13 by JacksonLv