NordicPlayground/nRF51-ble-bcast-mesh #87: How does a node with BLE_Gateway example actually rebroadcast data?

rifatmahmud, 2016-04-12:
When handle value is updated in mesh_gatt service, I see a call to [vh_local_update](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/src/mesh_gatt.c#L414). I assume, this broadcast the incoming data to mesh.

But, on main.c of BLE_Gateway example, I can't see any call to rbc_mesh_value_set on [new value or update value events](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/examples/BLE_Gateway/main.c#L119).

To rebroadcast the incoming data(updated via both gatt service or mesh itself), should I call the function [rbc_mesh_value_set](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/src/rbc_mesh.c#L213)  on new value and updated value events?

trond-snekvik, 2016-04-12:
The GATT-service sort of sneaks in the backdoor. The gatt-module works alongside rbc_mesh_value_set, but doesn't use it directly. This is to get an event back to the application that something happened, and still get the version++ thing done in vh_local_update. From the application's point of view, getting an update over gatt is the same as getting it over the mesh - it's already in the system, being retransmitted by the time you get the event - you don't have to do anything :) From the version-handler's point of view, the gatt-update is the same as getting an update from the application.

rifatmahmud, 2016-04-12:
Thanks for the response
Can you please point to the code where this is happening.(Rebroadcasting after update from mesh). I need to make some changes there?
Thank you again. :)

trond-snekvik, 2016-04-12:
Depending on which functionality you want to alter, the local update triggers two code paths:
First, the [cache-task to just store the new value, to make it eligible for retransmissions](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/src/version_handler.c#L428), then [the retransmit-code triggering on timeouts, to push the data to the radio](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/nRF51/rbc_mesh/src/version_handler.c#L340). The second code path does not differentiate between local updates and mesh updates - it just reads out timeout-values and packets from the cache and sends all packets that should have been retransmitted at the current timestamp (or earlier).

closed 2016-11-30 by trond-snekvik