NordicPlayground/nRF51-ble-bcast-mesh #88: mesh rebroadcast format

JacksonLv, 2016-04-14:
I am using the LightBlue to connect to a mesh node and send "00000003FF0000" to it .After that I use Packetsniffer and a dongle to capture its rebroadcast package.I saw it was "0A16E4FE0000xxxxFF0000" and I got the adva is 
0	1	2	3	4	5	6	7	8	9	10
0A	16	E4	FE	00	00	XX	XX	FF	00	00

as I know ,the first byte is the  adva package length -1, what I not not sure is the 6th and 7th ,it 's random??

JacksonLv, 2016-04-14:
I found it is the rebroadcast time ，6th is low byte,7th is high byte.

closed 2016-04-14 by JacksonLv

trond-snekvik, 2016-04-14:
You're close. Take a look at the [packet format diagram](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/sdk-8-support/docs/how_it_works.adoc#air-interface-packets). In fact, I recommend reading through that document, as it should give a nice overview of the inner workings of the mesh. The 6th and 7th bytes represent the version number of the data. Every time you set the data in the gatt server or via the API functions in rbc_mesh.h, the version number of that data is updated. Data is indexed by their handles (bytes 4 and 5). The goal in the mesh is that every device agree on the same data and version for every handle.