NordicPlayground/nRF51-ble-bcast-mesh #89: How to get all handles

czb, 2016-04-14:
Is there any way how to get all the handles that are stored on the device?

The idea is to advertise by NUS the fact the device received a new value from RBC. Mobile phone will by scanning detect the change and connect to the device. Then the mobile phone will request all the data from all the handles that are stored on the device to process it further.

Basically I want my mobile phone to be connected to the mesh and get updates from the mesh.

trond-snekvik, 2016-04-14:
Hi,
Currently, there's no fast way to read out all handles. You can of course make an attempt with `rbc_mesh_value_get()`, iterating over all the handles, but it'll be really slow, as the function has to search through the handle cache on each call. 

I think adding a `get_known_handles()` function is an interesting feature that could be useful, but I'd like to hear more about your use case first, to see if we can come up with a faster solution with the existing framework :wink: Is there any reason why you'd like to wait until detecting changes before connecting to a mesh device? If you're already connected by the time the updates come, you'll get them as a GATT characteristic notification - would this be a sufficient solution? 

You could also create a small app on the device which takes every `RBC_MESH_EVENT` coming in, and stores it in a list along with its data, then send that over your own GATT-service upon request.

czb, 2016-04-14:
Hi,

thanks for your quick response. The idea is to have a mesh consisting of nrf51 gateways, nrf51 sensors and one or more iPads. The sensors regularly advertise sensor measurements - lets say every 15 minutes. The sensor measurements spread through the mesh to all gateways. iPad is running all the time and periodically scans to see if any changes happened on the nearby nrf51 gateways. If so, it connects to them requests the senor measurements and disconnects again. I do not wish to be connected with iPad all the time because the battery would be drained very quickly.

Probably I will catch new value event in RBC and set handle number to my own array of handles. I just thought there might be any other way without duplicating the list of handles ;) .

trond-snekvik, 2016-04-14:
If you have a known set of handles (say handle 0 is kitchen temperature, handle 1 is bathroom humidity and so on), reading them out one by one over GATT shouldn't cause any significant delay. The only issue is if your handles are allocated randomly throughout the entire 65k handle range, without the iPad or gateway knowing which handles to search for.

I think I would solve it by having an "all-clear" handle controlled by the iPad: Each time you've connected and read out all handles, you push something to the designated "all-clear" handle (a timestamp, for instance). This spreads through the mesh, and all gateways stop advertising. As soon as a gateway detects new sensor updates, it starts advertising again, as a way to tell the iPad that it has new data.