NordicPlayground/nRF51-ble-bcast-mesh #93: Setting Imax and Imin

kentmartin99, 2016-04-18:
Hi,
We are having issues with long delays for messages to update (SDK V7)
I was wondering how to adjust the Imax, Imin and adv_int_ms.
Currently they are set to:->
	init_params.adv_int_ms = 50;
    trickle_setup(1000 * adv_int_ms,600, 3);
void trickle_setup(uint32_t i_min, uint32_t i_max, uint8_t k)
why is imin set higher than imax?
what is the significance of setting imin to 1000x adv_int_ms?
Is this 1000 value hard coded in other areas of code as well that would need to be changed
if this line is changed?
looks like imin is compared, and imin*imax is also compared.
so seems like min time is 50000ms(50s) and max time is 600*50000 (30,000S,8.3Hrs)
this can't be right? How do I interpret these two lines of code to determine the min and max transmit
latency times? How can I adjust these to suit a better latency for our system?

Thanks,
Kent.

trond-snekvik, 2016-04-18:
Hi Kent,

The Trickle standard has somewhat confusing naming of the two parameters. Imax is actually a multiplier for Imin, where the value you would expect to be Imax is actually `i_min * i_max`. We considered changing the naming scheme for our implementation, but decided to stick to the standard's definitions.

The 1000x in the adv_int_ms is to convert from ms to µs, which is the timestamp resolution in the mesh. It's hardcoded (and shouldn't be changed), but I think a #define would be a more elegant and readable way to go about it, as proven by the confusion it's creating. 

Your numbers have to be divided by 1000 to get the correct boundaries: min time is 50 ms, max time is 600 * 50000µs = 30 seconds.

trond-snekvik, 2016-12-01:
Closing due to inactivity. Reopen if there's any further confusion.

closed 2016-12-01 by trond-snekvik