NordicPlayground/nRF51-ble-bcast-mesh #94: Required RAM when RBC_MESH_SERIAL is defined

lopeznegrete, 2016-04-19:
According to the main README file this project should work with the nRF51822_QFAA rev 3 chip.

I have downloaded the last version of the project 0.8.3 and I have found several thins that confuse me about the aforementioned statement.

First of all, for some reason I do not understand I have to re-add 3 files (mesh_aci.c, serial_hanlder_spi.c and spi_slave.c) although the paths are correct at first.

Secondly, I have found that the IRAM1 memory area under project target is defined to be:

Start at: 0x20004000
Size: 0x4000

This means that the device is leaving 0x4000 for the SD and 0x4000 for the application code. That is a total of 32Kb of memory only available on the nRF51822 QFAC version.

If I modified the IRAM1 area as:

Start at: 0x20002000
Size: 0x2000

I am able to compile the version without serial communication (RBC_MESH_SERIAL not defined) whereas if RBC_MESH_SERIAL is defined I need to increase the RAM section.

Thus, my question is: Is it required to use the nRF51822_QFAC chip for the version with serial communication (RBC_MESH_SERIAL defined)?

I have not seen this explained on any of the READMEs section.

trond-snekvik, 2016-04-20:
Hi, 

The example projects are set up to target the 32k RAM/256k Flash by default, but I can see how the configuration is misleading, and I'll follow up on that. 

You can run the mesh (with serial) on the xxAA permutations of the nRF51 series, but it requires some tweaks to the various mesh parameters to fit in RAM. By default, the BLE_Gateway example compiles to 23k of flash and  8.3k of RAM. Most of the RAM is spent in various buffers, which are scaled up way beyond what a typical application needs. This is mostly to avoid confusing issues for those that are building applications on top of the examples, when they run out of space.

- The startup file arm_startup_nrf51.s declares a 2048 byte heap - the framework doesn't use it. You could also experiment with smaller stack size, but be careful.
- [rbc_mesh.h declares some cache sizes](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/rbc_mesh.h#L59). These are waaay too large for normal applications, and can be scaled down to about 10% for the example, and most basic applications. [More on caches here.](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/docs/how_it_works.adoc#value-storage)
- The app and radio event queues could be reduced from 16 to 4 entries, if you need even further reduction. I would keep the the internal event queue at 16.

All the queue length and cache size changes reflect on the packet pool size, which is the biggest memory hog. You may also experiment with a custom sized packet pool, but you could risk running out of packets if you go too low. With all of these reduced to a minimum, It's possible to compile the BLE_Gateway in the serial-config with 4k of RAM without optimization.

As for the RAM start address - that one's scaled extra high to make sure we can fit any additional services in the GATT server. You're right to set it to 0x20002000 for the example. 

I think your problem with the serial-files having to be readded should be solved by changing the Keil-target to nRF51822 xxAA Serial, from the drop down menu next to the configuration button.

lopeznegrete, 2016-04-20:
Thanks Trond. This is a comprehensive and very helpful answer.

closed 2016-04-20 by lopeznegrete