NordicPlayground/nRF51-ble-bcast-mesh #97: pstorage with mesh

JacksonLv, 2016-04-27:
Hi there,I am using the 8.1.0 SDK and I want to send data to nrf51 then save them into flash so that it can "Power down save".I test the pstorage library(Update 22.1.2016) [here](https://devzone.nordicsemi.com/question/15271/how-can-i-write-10kb-of-data-to-internal-flash/?answer=17300#post-id-17300)  worked fine . So I add mesh function to it ,Here is my code.
`int main(void)
{
    uint32_t err_code;
    
    // Initialize
    timers_init();
		uart_config();
    ble_stack_init();
//    bsp_module_init();
    scheduler_init();

	
    rbc_mesh_init_params_t init_params;

    init_params.access_addr = RBC_MESH_ACCESS_ADDRESS_BLE_ADV;
    init_params.interval_min_ms = MESH_INTERVAL_MIN_MS;
    init_params.channel = MESH_CHANNEL;
    init_params.lfclksrc = MESH_CLOCK_SOURCE;
    
    uint32_t error_code = rbc_mesh_init(init_params);
    APP_ERROR_CHECK(error_code);

    /* init BLE gateway softdevice application: */
    nrf_adv_conn_init();

    NRF_GPIO->OUTCLR = (1 << 4);
		

    /* fetch events */
    rbc_mesh_event_t evt;
	
		pstorage_test_store_and_update();
	
    // Enter main loop
    for (;;)
    {
				if (rbc_mesh_event_get(&evt) == NRF_SUCCESS)
        {
            //rbc_mesh_event_handler(&evt);
            rbc_mesh_packet_release(evt.data);
        }
        
        sd_app_evt_wait();
        app_sched_execute();
        power_manage();
    }
}`

`static void ble_stack_init(void)
{
    uint32_t err_code;

    // Initialize the SoftDevice handler module.
    SOFTDEVICE_HANDLER_INIT(MESH_CLOCK_SOURCE, NULL);
	    // Initialize the SoftDevice handler module.
    //SOFTDEVICE_HANDLER_INIT(NRF_CLOCK_LFCLKSRC_RC_250_PPM_4000MS_CALIBRATION, NULL);

#if defined(S110) || defined(S130)
    // Enable BLE stack 
    ble_enable_params_t ble_enable_params;
    memset(&ble_enable_params, 0, sizeof(ble_enable_params));
#ifdef S130
    ble_enable_params.gatts_enable_params.attr_tab_size   = BLE_GATTS_ATTR_TAB_SIZE_DEFAULT;
#endif
    ble_enable_params.gatts_enable_params.service_changed = IS_SRVC_CHANGED_CHARACT_PRESENT;
    err_code = sd_ble_enable(&ble_enable_params);
    APP_ERROR_CHECK(err_code);
#endif
    
    // Register with the SoftDevice handler module for BLE events.
		//err_code = softdevice_ble_evt_handler_set(ble_evt_dispatch);
    err_code = softdevice_ble_evt_handler_set(sd_ble_evt_handler);
    APP_ERROR_CHECK(err_code);
    
    // Register with the SoftDevice handler module for BLE events.
		err_code = softdevice_sys_evt_handler_set(rbc_mesh_sd_evt_handler);
    APP_ERROR_CHECK(err_code);
				err_code = softdevice_sys_evt_handler_set(sys_evt_dispatch);
		APP_ERROR_CHECK(err_code);
}`

`static void sys_evt_dispatch(uint32_t sys_evt)
{
  
		pstorage_sys_event_handler(sys_evt);
}`

I test the mesh worked fine but the pstorage get error `pstorage CLEAR ERROR callback received `

trond-snekvik, 2016-05-02:
Hi, 

There is a bit of friction between pstorage and the Softdevice timeslot system. The pstorage module calls the flash API through the Softdevice, as the Softdevice "owns" that peripheral. The Softdevice can only use the flash API when there is no timeslot running, but unfortunately, the mesh attempts to allocate as much time as possible from the Softdevice. Each time you call the flash API (this happens a lot in pstorage, of course:) ), it has to wait for the current timeslot to finish. This can take up to 10 seconds, and by that time, the Softdevice has freaked out, and your flash-operation fails. This gives you your clear-error in the callback. 

To fix this, you should attempt to restart the timeslot after each successful call to an `sd_flash_*` function. There are several in pstorage.c, and each call will wait for an available timeslot. To restart the timeslot, call `timeslot_restart()` in timeslot_handler.h in _/nRF51/rbc_mesh/include_.

JacksonLv, 2016-05-04:
ok,thanks,I will try it

closed 2016-05-04 by JacksonLv