NordicPlayground/nRF51-ble-bcast-mesh #100: Inhibiting continious flooding?

victorpasse, 2016-04-29:
Hi, If I want a node to not repeat messages that has i_relative over g_i_min*g_i_max, How would I do that?

trond-snekvik, 2016-05-02:
That could be achieved by stopping transmissions on the Nth TX-event for that specific handle. Knowing when the interval reaches the ceiling is not trivial with the current API, and if absolute accuracy is required, you'll have to make some minor additions to the framework code: 

The Trickle interval is updated on every call to [`trickle_rx_consistent()`](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/trickle.c#L118) and [`trickle_tx_timeout()`](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/trickle.c#L158). These functions are called exclusively from the  handle storage ([for TX](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/trickle.c#L158) and [for RX](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/handle_storage.c#L609)). After these two calls, you can check their trickle intervals against the max:

```
if (m_data_cache[data_index].trickle.i_relative == MESH_TRICKLE_I_MAX * trickle_min_interval_us)
{
    trickle_disable(&m_data_cache[data_index].trickle);
}
```

Note that the `trickle_min_interval_us` is not currently stored in this module, but it's passed into the `handle_storage_init()` function, where you can store a static global version of it.

victorpasse, 2016-05-02:
Thanks!
This disabled the trickle and it never seems to come online after, Do I need to modify something else?

trond-snekvik, 2016-05-02:
Ah, yep. We would want to reenable trickle when we push new data to the handle. Pushing new data exclusively happens through the [`handle_storage_info_set()` function](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/handle_storage.c#L363). At [line 396](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/handle_storage.c#L396), we reset the trickle instance. Before that, we want to re-enable it as well. Do this right before the call to `trickle_timer_reset()`:

```
trickle_enable(&m_data_cache[data_index].trickle);
```

victorpasse, 2016-05-02:
Thank you verry much!

victorpasse, 2016-05-19:
I also had to add
trickle_enable(trickle);
in trickle_rx_inconsistent
before 
trickle_timer_reset

Due to the following situation:
For handle X:
node 1 has version 100
node 2 has just powered on and changes handle X, now the sent version is 1
node 1 receives version 1 and does not send version 100 back to 1 as the trickle is disabled.

Now 100 packages has to be sent before node 1 "listens" to node 2

victorpasse, 2016-05-19:
Should I always do 
trickle_enable
before every call to 
trickle_timer_reset?

trond-snekvik, 2016-05-20:
yes.