NordicPlayground/nRF51-ble-bcast-mesh #102: rbc_mesh_stop not working

czb, 2016-05-02:
I would like to initialize rbc in stopped state so I do:

	err_code = rbc_mesh_init(init_params);
	APP_ERROR_CHECK(err_code);
	
	err_code = rbc_mesh_stop();
	APP_ERROR_CHECK(err_code);

However when I measure current it looks like the rbc mesh is actually not stopped. Can I call rbc_mesh_stop just after rbc_mesh_init?

czb, 2016-05-02:
I have rewritten my code to bypass the issue. Now the problem I face is that from time to time, although I call rbc_mesh_stop from app_timer the mesh does not actually get stopped. Is there any way how to make sure that the mesh will stop?

trond-snekvik, 2016-05-02:
Hi, I wrote a hotfix for your issue, as it was caused by a bug which required more than a couple of lines to fix. You can find it here: https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/tree/hotfix_issue_102. The branch should address both stop right after init and odd stop behavior, could you help us verify this?

czb, 2016-05-04:
I have tested the hotfix for the last 24 hours and it seems to work fine. Thanks a lot!

closed 2016-05-04 by czb