NordicPlayground/nRF51-ble-bcast-mesh #104: Notification interference

vliedel, 2016-05-06:
Hi, I'm sending out 11 notifications per second and saw that when I enable meshing, I'm missing a notification pretty regularly (every 6th or 7th notification or so).
Is this simply due to the increased radio activity or could it be a bug?

[image]
In this graph, each peak at 10 means one notification is missed, peak of 19 means 2 notifications are missed. Horizontal axis: each 9 is one received notification.
Without meshing enabled, this graph was a flat line, no notifications missed.

trond-snekvik, 2016-05-09:
Just to make sure: you're talking about GATT notifications, not [radio-notifications](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s110.api.v8.0.0/group___n_r_f___s_o_c___f_u_n_c_t_i_o_n_s.html?cp=2_6_0_0_0_2_6_3_43#ga15378609daca09588e620b64555fc2e8), right?

What is your connection interval for this? Typically, you wouldn't be able to send more than one notification per connection event, and the mesh will be sending quite a few. You can monitor the mesh GATT activity over [here](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/mesh_gatt.c#L171), but I don't have any quick fixes except shrinking your connection interval. You could disable [acks](https://github.com/NordicSemiconductor/nRF51-ble-bcast-mesh/blob/master/nRF51/rbc_mesh/src/mesh_gatt.c#L174) for the commands, but that would only help you if you are pushing commands from the master. You could also choose to stop the mesh notifications all togheter, and poll instead, but you'll reduce functionality dramatically.

daviddedwin, 2016-05-09:
I do not understand how we miss a Handle Value Notification on the softdevice, since the BTLE link layer is a reliable link. _Please clarify_. I am assuming "radio-notifications".

vliedel, 2016-05-09:
I meant GATT notifications, but I just found out I might be doing something wrong, so I'll first test a bit more. Thanks for the feedback.

vliedel, 2016-05-12:
So, I now check if a notification is sent, before sending the next. Then, instead of missing notifications, my send buffer overflows quickly with mesh enabled, without meshing this doesn't happen.

trond-snekvik, 2016-05-12:
You can try reducing the mesh GATT traffic, as I described above, or increase the size of your send buffer. Reducing the connection interval should improve your notification throughput, but stay above 15ms to give the mesh some room to breathe! These are the only options you have at this point, as the limit you're hitting now is a hard one - there simply isn't enough bandwidth in GATT to support your traffic rates.

vliedel, 2016-05-12:
Ok thanks, it's always good to know the limits :)

daviddedwin, 2016-05-12:
Are you waiting for the BLE_EVT_TX_COMPLETE for the Notification before sending the next one ?

vliedel, 2016-05-12:
Yep

closed 2016-11-30 by trond-snekvik