NordicPlayground/nRF51-ble-bcast-mesh #106: Generating Random Number in rbc mesh

rifatmahmud, 2016-05-16:
How to generate a random number in nRF51822?
I am afraid to use the rng example in peripheral of SDK, thinking it might mess up the mesh.

daviddedwin, 2016-05-16:
If you want a random number you can get it from the softdevice call to the RNG hardware.
See https://devzone.nordicsemi.com/question/7244/random-pool-numbers-documentation/ for a bit more detail. Will that work for you ?

rifatmahmud, 2016-05-16:
Ok, but just wondering, what rand library in rbc mesh is for?
Can I use that, getting the seed from timer or rtc?

trond-snekvik, 2016-05-16:
You can seed the prng with rand_prng_seed(), which seeds from the SD-pool (hardware rng). The prng is based on [Bob Jenkins' small PRNG](http://burtleburtle.net/bob/rand/smallprng.html), which has decent cycles and uniformness, but it's not suited for cryptographic purposes. You can also use the `rand_hw_rng_get()` for accessing the sd_rand, but you could also use the SD interface directly.

closed 2016-07-28 by rifatmahmud